toblum/McLighting #1: Does this work with a ESP8266-01?

drschnalli, 2016-10-24:
I got the Software up and running on my ESP8266-01 but i don't know which pin to use as data pin for the ws2812 strip. I set PIN 1 in the config.

toblum, 2016-12-27:
Dear drschnalli,
I missed that you opened an issue. According this (http://www.instructables.com/id/How-to-use-the-ESP8266-01-pins/) instructable, the ESP-01 has GPIO 0 und GPIO 2 available. I think you should be able to use one of those.

closed 2016-12-27 by toblum

dunk8888, 2017-11-16:
Hi,ime trying to use esp_01 but ime getting nothing out via serial? any ides?

dunk8888, 2017-11-16:
@drschnalli Hi did you get your esp_01 working??

toblum, 2019-01-31:
Hi @peexiz,

ths sketch may be too big for a standard ESP01 (512k) without tweaks, a ESP01S may work. There are instructions here: https://github.com/toblum/McLighting/wiki/Software-installation#setup-the-arduino-ide

Regards
Tobias

debsahu, 2019-01-31:
Alternatively, try the develop branch. And uncomment `//#define USE_HTML_MIN_GZ` to see if everything works for you.