toblum/McLighting #2: I have integrated FastLED into your project (among other things)

russp81, 2016-11-05:
See here:  https://github.com/russp81/LEDLAMP_FASTLEDs

You did awesome work man, I just like using FastLED better.  Mostly due to my limited skills, and it seems pretty powerful.

Just wanted to let you know...

jake-b, 2016-12-02:
I have forked russp81's fork at https://github.com/jake-b/Griswold-LED-Controller for those who are interested.  Some code clean-up and improvements.  I intend to use it stand-alone to control my Christmas lights this year.

toblum, 2016-12-27:
Hi @russp81 and @jake-b ,
I wasn't notified about this issue, so I missed your messages too. Using FastLED was also on my list, so I will definitively have a look into your forked projects. 
Best regards
Tobias

closed 2016-12-27 by toblum