toblum/McLighting #3: ESP not reachable after connecting to WIFI

nuovocampo, 2016-11-20:
Hi Tobi,

First of all, awesome project! Thanks a lot for sharing :)

Unfortunately, I cannot reach the ESP after it connected to the WIFI. The first start was flawless and I got redirected to the WIFI config page where I entered my SSID & PW. 
`*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.2.126
connected...yeey :)
Open http://ESP8266_01.local/edit to see the file browser
`
After the ESP successfully connected to my WIFI network but from thereon is not reachable at all.
I can see it on my router as a connected device but can't reach the ESP8266_01.local page nor get even a ping back...
Any idea what might be the issue? I'm guessing it must be sth with the webserver but really don't know where or how to look :-/

Many thanks!
Alex

jalil407, 2016-11-29:
Http:// ESP8266_01.local work only with apple or Mac,they have bonjour service

jalil407, 2016-11-29:
From android device you can get mclighting only with ip adress

jalil407, 2016-11-29:
From Windows if you want to open ESP8266_01.local, you must install bonjour service,it work fine

jalil407, 2016-11-29:
To avoid searching ip adress from your router, install app Bonjour from playstore, bonjour app search bonjour service in your home network,and display the ip adress

nitin4212, 2016-12-27:
hi sir problem in mc lighting first of all when redirect 192.168.4.1 . how to config static ip ,DNS, Gateway .pl help me as soon as possible.
[image]

toblum, 2016-12-27:
Dear @nuovocampo,
I hope the problem was solved with the help of @jalil407 .

@nitin4212:
McLighting is designed to work with DHCP and dynamic IP adresses, so no need to configure static IP, DNS and gateway. If you need a static IP, you may have the possibility to make the dynamic IP "sticky" in your routers configuration, so that the ESP will always get the same IP from your router.
McLighting uses the great WifiManager library to handle WLAN management. You can have a look at the documentation there: https://github.com/tzapu/WiFiManager#custom-ip-configuration

Best regards
Tobias

nitin4212, 2016-12-28:
Thanks Sir

nitin4212, 2016-12-28:
how can i control multiple light through mclight, like Philips hue

toblum, 2016-12-29:
McLighting does currently not support multiple lights with one webpage. But you could install multiple lights and control each with it's own webpage.

pallavi199427, 2017-12-21:
wifi is not configurable on android it saying file not found when i am trying to connect to given ip

toblum, 2017-12-22:
@pallavi199427
Please follow the wiki until the last step. https://github.com/toblum/McLighting/wiki/Software-installation

pallavi199427, 2017-12-23:
Hey thanks for replying I just wanted to know how do I upload the files in
coz it saying file nt found

toblum, 2017-12-23:
Currently McLighting only with in station mode, not with own access point.

pallavi199427, 2017-12-26:
okey can i add my custom led pattern in this and where should i modify?? thanks in advance

toblum, 2017-12-26:
McLIghting relies on the great WS2812FX library, You could implement your own patterns by modifying that code.