toblum/McLighting #7: HTTP REST API /set_switch

fthmhmt, 2017-01-05:
Thanks for the great Tutorial!

Is it possible that you add some more options to HTTP Rest API? I am using OpenHAB 2 and could run just the /set_brightness of the API.

Therefore /set_switch would be interesting to switch on and off. Its possible to bring the brightness to 0 but this is not possible by buttons.

toblum, 2017-01-06:
Dear @fthmhmt,

extending the API is easy and could be done directly in McLighting.ino.

But set_switch is only a shortcut for alerdy existing API endpoints, I think:
/set_switch=1 is equal to /set_brightness?c=255
and
/set_switch=0 is equal to /set_brightness?c=0 

Isn't it possible to use these API calls in OpenHAB?

Best regards
Tobias

fthmhmt, 2017-01-07:
Got it working with the settings below

**rules/mqtt.rules**
```
rule "Neopixel"
    when
        Item NeoPixel received command
    then
        switch(receivedCommand) {
            case ON  : sendHttpGetRequest("http://192.168.xxx.xxx/set_brightness?p=100")
            case OFF : sendHttpGetRequest("http://192.168.xxx.xxx/set_brightness?p=0")
        }
end
```

**items/mqtt.items**
`Dimmer NeoPixel_Dimmer "Neopixel Dimmer" <light> ["Lighting"] { http=">[*:POST:http://192.168.xxx.xxx/set_brightness?p=%2$s]" }`

**sitemaps/mqtt.sitemaps**
```
Frame label="NeoPixel Dimmer" {
        Slider item=NeoPixel_Dimmer switchSupport
        Switch item=NeoPixel_Dimmer
}
```

toblum, 2017-01-07:
Dear @fthmhmt,

great that you figured it out. 

I don't have any SK6812 available here, so I don't know what exaclty to do, to make them work. If the neopixel library natively supports them, it could be sufficient to enhance the API endpoints with an additional parameter to control the warm white led, But I'm not sure.

Best regards
Tobias

closed 2017-01-07 by toblum