toblum/McLighting #8: Is this a mistake?

ssolmaz, 2017-02-09:
Hi Tobias,

I like v2. Thank you for this project.
I added homebridge. I use it via http without any problems.
But I have a problem via homebridge. I select color. 
When I want to adjust the brightness, it flashing at every level (once).
and the current color is light color. not dimmed, the color is light color.
via http brightness setting decreases the light. 
you can see the serial print output below.
I want to reduce the light via homebridge.
also will it be mqtt support?

[image]

toblum, 2017-02-09:
Dear @ssolmaz,

this looks unexpected, but I have currently no homebridge installation available to test that out, because my raspberrys SD card died and i have to reinstall. When I have a working system again, I will have a look at it. In McLighting v1 it worked, in v2 I did not try it out so far.

I did not find time to look into MQTT support right now, but maybe one day...

ssolmaz, 2017-02-10:
Thank you for your answer Tobias.
I did a clean setup last night (different nodemcu).
There is no problem with color dimming via homebridge (home app).
It's just flashing when dimming with the home app. but via eve app no flashing.
I hope to help for the solution. I can shoot video if you want.
Also mqtt support is absolutely necessary.
Because the strip does not open that is closed color and brightness settings.

toblum, 2017-02-11:
I tried that out today. the flashing seems to come from the app. It sends these extra requests to simulate a "dimming" effect, I think. There seems to be no way to deactivate this.

closed 2017-02-11 by toblum

dunk8888, 2017-11-16:
Hi,ime having trouble setting this up,ime getting the led flashing on my esp-01 the strip is blue,and nothing on serial,also the esp doesnt ome up as the name i defined,its coming up esp3238876,any ideas it compiles ok,ive tryed all my esp8266 modules none seem to work,ime sure there has to be somthing in serial to show what ip to use??

toblum, 2017-11-18:
I haven't tried this on ESP-01 for a while because I don't have any available anymore and I find the NodeMCU boards a bit more handy. 
Maybe memory is getting short on the ESP-01. Did you try out to deactivate MQTT/OTA?