toblum/McLighting #9: Homebridge/Home Assistant Switch & get_switch parameter

markbajaj, 2017-02-10:
HI Tobias,

Thanks for all your work! Great update. 

I am having a small issue with the http://<IP of ESP>/get_switch. Basically it always returns a 1, whether the LEDs are on or off. Is this because the /off command just changes the colour to RGB 0 0 0 ?

I have tried turning the brightness to 0, RGB to 0 and mode to 0 and the return is still 1. The issue here is that I have added it to home bridge and the status always shows on. I also use Home Assistant and have added a command line switch, again the status is always on so I have had to remove the status line.Below is a snippet from my Homebridge code:

                {
                        "accessory": "HTTP-RGB",
                        "name": "strip",

                        "switch": {
                                "status": "http://IP/get_switch",
                                "powerOn": "http://IP/all?r=255&g=255&b=255",
                                "powerOff": "http://IP/off"
                        },

                        "brightness": {
                                "status": "http://IP/get_brightness",
                                "url": "http://IP/set_brightness?p=%s"
                        },

                        "color": {
                                "status": "http://IP/get_color",
                                "url": "http://IP/all?rgb=%s",
                                "brightness": true
                        }
                }

Here is my command line switch in Home Assistant:

  - platform: command_line
    switches:
      light_strip:
        command_on: "/usr/bin/curl -X GET http://192.168.101.161/all?r=255&g=255&b=255"
        command_off: "/usr/bin/curl -X GET http://192.168.101.161/off"
#        command_state: "/usr/bin/curl -X GET http://192.168.101.161/get_switch"

Do you have any ideas on how I can report the status?

Thanks.

markbajaj, 2017-02-10:
Perfect!!! worked a treat!!!!! Thanks.

closed 2017-02-10 by markbajaj

toblum, 2017-02-10:
Fixed in: https://github.com/toblum/McLighting/commit/33804de86e01e339afa9451fd6fb0ed73de149c1

markbajaj, 2017-02-10:
> Dear @markbajaj <https://github.com/markbajaj>,
> I think I'll have to look into this, but as a quick fix you could try to comment out that line: https://github.com/toblum/McLighting/blob/master/Arduino/McLighting/McLighting.ino#L363 <https://github.com/toblum/McLighting/blob/master/Arduino/McLighting/McLighting.ino#L363>
> That should fix the problem for now.

Thanks for fixing it :)

I also have added code for OTA updates in the Arduino IDE - where would I share those?

Thanks

Mark

toblum, 2017-02-11:
@markbajaj Having OTA support would be great. You could file this as a pull request in Github. That would be great.