toblum/McLighting #12: McLighting Artnet Branch

drschnalli, 2017-03-28:
Hello, i just realized that there's a ArtNet Branch of your good work! It is working but i got a problem when switching back from ArtNet to another Preset. 

This Problem does not occur when i first close the ArtNet Connection and then switch the preset.

Does anyone have this problem, too? Do i first have to end the ArtNet Session before switching the presets? 

Forgot to mention what happens: when i forgot closing the artnet connection and then switch to another preset the esp8266 crashes and does not respond to pings and commands, when i reboot it everything is fine ok.

toblum, 2017-03-29:
Hi @drschnalli,

the Artnet was donated by another user and I just integrated it into the branch but did not find time to play with it. As I have no matrix build myself I just tried a little bit with it but did not dig deeper so far.

I'm also trying to contact the original author of the script to find out if it's OK to include his work into McLighting, but got answer so far.

Regarding your problems with this version, you can try to ask in the gitter group https://gitter.im/mclighting/Lobby as there are some artnet users out.

Regarding your question with the different start mode:
This should be easy. Just set mode to SET_MODE instead of RAINBOW in line 19 of definitions.h (https://github.com/toblum/McLighting/blob/master/Arduino/McLighting/definitions.h#L19). And set ws2812fx_mode to the number of the ws2812fx effect.

Best regards
Tobias

chrifey, 2017-04-23:
Hi @drschnalli,

I can reproduce the issue, but did not have a deeper look into it since my skills are not that big  (I just started a month ago).

My main goal was to implement a TPM2NET -> WS2812b node. I also found the work of mcnobby, but actually I find TPM2NET better since it seems to work easier and the Tools like Jinx! work like a charm, even with bigger matrix sizes. With artnet one needs to deal with the universes and build some magic to have more than 512 channels on one pin.

Since I find this project very nice (so thanks to @toblum) I would find it very nice if I could try to implent the code from here:

closed 2018-01-13 by toblum

toblum, 2018-04-14:
HI @chrifey,

I just saw your message. I you have created something that works we could work together to integrate it into the main McLighting branch. Unfortunately I don't have any experiences with TPM2NET.

Regards
Tobias

retradnews, 2018-04-14:
I would also prefer tmp2net. This would be a really great Feature (not only for my Installation, I think). Just because the Audio Support ( with out needs for a Matrix, you can also add great patterns with stripes).
Unfortunately i'm not so deep In  programming that i could be a great help at this.

chrifey, 2018-04-14:
Hi,

thanks for coming back. I was not active with this quite for a while. In the past I went with the code mentioned in my first post since that was sufficient (but not very handy).

I just searched a bit and found the following project: [https://hackaday.io/project/25851-pixel-led-wifi-controller-tpm2net#menu-description](https://hackaday.io/project/25851-pixel-led-wifi-controller-tpm2net#menu-description)

I think that looks promising. Will try this standalone and maybe we could try to add this to your project. tpm2.net seems to be quite easy to implement from a protocol perspective.

Regards
Christian