toblum/McLighting #15: Unable to compile after all libraries are installed.

Fireflaker, 2017-04-16:
In file included from C:\Users\win7\Desktop\EE\WS2812LED\McLighting-master\Arduino\McLighting\McLighting.ino:97:0:

sketch\request_handlers.h: In function 'char* listStatusJSON()':

sketch\request_handlers.h:84:8: warning: address of local variable 'json' returned [-Wreturn-local-addr]

   char json[255];

        ^

C:\Users\win7\Documents\Arduino\libraries\WebSockets\src\WebSocketsClient.cpp: In member function 'void WebSocketsClient::begin(const char*, uint16_t, const char*, const char*)':

C:\Users\win7\Documents\Arduino\libraries\WebSockets\src\WebSocketsClient.cpp:69:16: error: 'RANDOM_REG32' was not declared in this scope

     randomSeed(RANDOM_REG32);

                ^

What does it mean? How can I solve it? 
Is "RANDOM" function not available?
Please help. Thank you.

toblum, 2017-04-16:
Hi @Fireflaker,

these error messages seem to be related with older versions of the Arduino core: https://goo.gl/Ktw1QY. What version do you use? Is there a update available in the board manager?

Regards
Tobias

Fireflaker, 2017-05-04:
Thank you so much. Problem solved.

toblum, 2017-05-04:
You're welcome

closed 2017-05-04 by toblum