toblum/McLighting #19: Add called command in Websocket response?

codmpm, 2017-05-11:
It would be nice to have the issued command in the response of the websocket message. Something like `#ok` or `color:ok`.

Right now it kind of difficult to distinguish between an answer to a command or if you just polled for the current status. Or did I miss something?

Cheers,
Patrik

toblum, 2017-05-17:
Hi Patrik,

This sounds useful. I will have a look at it as soon as I find time. Or you could file a pull request if you already have something. 

Regards 
Tobias

codmpm, 2017-05-17:
Sadly not. Did not do a deep dive into code, yet ;-)

I have some more question, maybe you can get back to me. Alreay tried to reach out to you via FB.

Cheers.

codmpm, 2017-05-19:
resent :)

closed 2018-04-07 by toblum