toblum/McLighting #21: Does not go into AP mode / No Debug Output

ZeeZeeEff, 2017-05-21:
Hey,

I do have a problem with your software. When I flash the sketch, the LED's start in rainbow mode but there is no debug output in the serial monitor and the esp does not open an AP. What could be wrong?

Arduino IDE: 1.8.2
all other Libraries are on the same version that you've posted

ZeeZeeEff, 2017-05-21:
Okay, the issue can be closed!
I solved it by erasing the flash of the esp via esptool

ZeeZeeEff, 2017-05-22:
I got it to work now, but there is still no Debug output, strange
it is just outputting single blank's

toblum, 2017-05-26:
Good to hear that it works now. I see from the Gitter chat that debug also works now after erasing flash.

closed 2017-05-26 by toblum

dimiHH, 2017-11-01:
Hi @ZeeZeeEff , have you figured out what to do about the Serial output? I have the same problem - everything works (ap point, connecting to Wifi, uploading the html file and the webinterface, but I don't get sierial Output. 

I tried different baud rates too.

Thanks,
dimi

mogyoros, 2017-12-31:
@dimiHH 
For me commenting out the line in McLightning.ino solved the no serial debug issue.
`//  pinMode(BUILTIN_LED, OUTPUT);
`
Cheers,
Tamas

toblum, 2017-12-31:
Thanks @mogyoros,
yes, that would explain the problem. I wasn't aware of that conflict described here:
https://github.com/esp8266/Arduino/issues/2192

Regards
Tobias