toblum/McLighting #22: Enhancement Request: MQTT Control

noisufnoc, 2017-05-25:
Hello,

Thank you for this project, I have been trying to reliably control a NeoPixel with an ESP8266 and you made it very easy.  Ultimately I would like to control this light via my home automation platform [Home Assistant](https://home-assistant.io/).  MQTT control of the lights would make this integration very easy.

Thoughts?

toblum, 2017-05-26:
Hi,

I think there is already a way to implement this with Home Assistant using the REST API, but I don't know HomeAssisant in detail. Nevertheless MQTT would be nice to have and I definitely will look into this when I find time. The PubSub library uses some additional RAM and memory, so I will have to try out if it's working  with all libraries enabled.

If you or someone else will do the first step on the integration I would definitely integrate this into McLighting.

Regards
Tobias

thundergreen, 2017-06-18:
+1 for home assistant integration with MQTT. I will maybe buy another esp to test this :)

LeonVos, 2017-06-21:
i just added MQTT to MCLighting.ino.
Page is in request

https://github.com/toblum/McLighting/compare/master...LeonVos:patch-1

thundergreen, 2017-06-22:
Awesome! It's already in master branch ? Wanna check this out

thundergreen, 2017-06-22:
> master...LeonVos:patch-1

Would it be possible to add a short manual how to integrate those in home
assistant ?

thundergreen, 2017-06-22:
I overflew the file and couldn't locate mqtt username and password . Neither topic .I might have read a bit too fast and on the phone but I was just wondering if I miss something ?

LeonVos, 2017-06-22:
User&Passw is not needed for this  , topics are at the end  (ws2812-set/mode,  ws2812-set/state )

Remind you that the control for mode does not work yet.

I use io-broker, for H.A..  integration search for " mqtt home assistant" lots off sources for it.

thundergreen, 2017-06-22:
when u talk about mode. are those effects?

toblum, 2017-06-23:
@LeonVos Great! I will have a look at it as soon as I'm back from holidays and find time.

Could you file this as a pull request in the McLIghting repo? Then I will create a standalone branch where we can test and do enhancements.

Regards

toblum, 2017-08-06:
Plese have a look here: https://github.com/toblum/McLighting/issues/26#issuecomment-320534925

nimbl, 2017-08-09:
First off, thanks for the great repo. I have two things to mention in reference to this new MQTT functionality that was added to the master branch.

1) I suggest refactoring the infinite mqtt connection loop to give up after a few retries. Right now, if the mqtt connection isn't established, then the webserver and websocket server will hang.  At: https://github.com/toblum/McLighting/blob/88b476db51211eb38807ab9b33676afcabd5ab65/Arduino/McLighting/McLighting.ino#L542 I suggest something like this instead:
`
if (!mqtt_client.connected()) {
 if(mqtt_reconnect_retries < mqtt_reconnect_retries_max){
  mqtt_reconnect();
 }
}else{
 mqtt_client.loop();
}
`
2) As a hint as to why large messages aren't working in pubsubclient, as mentioned in this todo: https://github.com/toblum/McLighting/blob/88b476db51211eb38807ab9b33676afcabd5ab65/Arduino/McLighting/request_handlers.h#L369

It seems that the pubsubclient doesn't respect redefining of MQTT_MAX_PACKET_SIZE, as you have in definitions.h. So any message over 128 bytes won't be sent. You can change this as a build option if using PlatformIO, but not with the Arduino IDE. See this issue https://github.com/knolleary/pubsubclient/issues/110

Hope this info helps, and I'm not posting this into the wrong place. Thanks for the excellent software.

toblum, 2017-08-11:
Thank you @nimbl.  I will have a look at it as soon I find time.

thundergreen, 2017-08-11:
why do i get 

```
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
```
and it just wont connect and how is the wifimanager supposed to work? which address do i have to type in?

LeonVos, 2017-08-12:
Hi, you need to connect your wifi with the esp (ssid looks like "espxxxx") default ip is 192.168.1.4

toblum, 2017-08-12:
IP should be: 192.168.4.1 I think
@thundergreen It Looks like your Board has stored valid credentials. Maybe you used it with wifimanager before. 
You could uncomment the line with wifimanager.resetsettings(). In McLighting.ino. Then run one time to reset everything. Then comment it again.

thundergreen, 2017-08-14:
how would i reset it then ?  will try this again now

thundergreen, 2017-08-14:
i reset it by uncommenting .. commted it again and it runs as accesspoint now .. strange

toblum, 2017-08-19:
Dear @thundergreen,
I hope everything is OK now. I will close the issue. If there are still problems, please just open a new issue.
Regards
Tobias

closed 2017-08-19 by toblum

toblum, 2017-08-19:
@nimbl 
As proposed by you, I added a limit to the MQTT connections tries: https://github.com/toblum/McLighting/commit/91571eafccb329ec1e7c96320c9b7a974bb426ea