toblum/McLighting #23: #include<core_esp8266_features>

gunjanmx4, 2017-06-08:
how to solve it 
[image]

toblum, 2017-06-09:
Hi,

which version of the ESP8266 core for Arduino do you use? Stable or latest?
See: https://github.com/toblum/McLighting/wiki/Software-installation#esp8266-board-support-for-arduino-ide

Regards

gunjanmx4, 2017-06-09:
> installation#esp8266-board-support-for-arduino-ide

thank you sir for your reply sir i am usind esp8266 12-e nodemcu model

toblum, 2017-06-09:
Dear @gunjanmx4,
thanks. But I was asking for the software version of the ESP Arduino core. Like shown in the attached link.
https://github.com/toblum/McLighting/wiki/Software-installation#esp8266-board-support-for-arduino-ide

Regards

closed 2017-08-06 by toblum