toblum/McLighting #24: Request: RGBW Mode

semaf, 2017-06-26:
Hey,

Is it possible to support RGBW Leds like SK6812 Strips?

toblum, 2017-06-27:
I think this may be possible as the Adafruit library seems to support these, at least in RGB mode. I don't know how the W can be adressed. But maybe you can give it a try if you have a SK6812 at hand.

Regards
Tobias

semaf, 2017-06-27:
As you talk to RGB with each byte, you have to talk to the 4th byte too which is seperate white to rgb. 

I have tested it over Homebridge with Homekit and the 4th byte, the W was lighten on. But your Arduino and index.htm can not handle with it, the brightness lower or higher or on off of the 4th byte. I did not study your complete code, just uploaded and tested ;P Need time to check it and give a test but could take time as I am not free.

codmpm, 2017-12-30:
Just to update this, McLightning works with RGBW Pixels but without using the white-channel.

Simply change the pixel type in the WS2812FX init from 

    WS2812FX strip = WS2812FX(NUMLEDS, PIN, NEO_GRB + NEO_KHZ800);

to 
    WS2812FX strip = WS2812FX(NUMLEDS, PIN, NEO_GRBW + NEO_KHZ800);

closed 2018-01-13 by toblum

martinschki, 2018-03-30:
Hi Tobias,
any plans to implement this?
I would be really interested for having a combined "normal" lamp with RGB options for the kids rooms
Thanks, ESP boards are on the way...;)

toblum, 2018-03-30:
HI @martinschki,

there is already a fork of McLighting that provides support for RGBW. Unfortunately it's not so easy to integrate into the main code, but you could give it a try: https://github.com/toblum/McLighting/issues/94

Best regards
Tobias

SirJMD, 2019-02-23:
Any news on this? I think McLighthing works really well, but lacking support for RGBW is a real bummer.

Support for RGBW is the difference between it being a fun project, and then being actually useful in a home.

ChristophA, 2019-02-23:
Have look here: 
https://github.com/FabLab-Luenen/McLighting
Unfortunately it‘s a separate effort, but I still hope that it will be integrated into the „master“ at some point in time.

SirJMD, 2019-02-23:
I've been looking at that one, but as you say it's a separate effort - which isn't necessarily bad, but I would have preferred a joint effort :)

VeniceNerd, 2019-07-10:
> Just to update this, McLightning works with RGBW Pixels but without using the white-channel.
> 
> Simply change the pixel type in the WS2812FX init from
> 
> ```
> WS2812FX strip = WS2812FX(NUMLEDS, PIN, NEO_GRB + NEO_KHZ800);
> ```
> 
> to
> 
> ```
> WS2812FX strip = WS2812FX(NUMLEDS, PIN, NEO_GRBW + NEO_KHZ800);
> ```

Where do I find this file that needs the change? Also, has official W support ever been fully integrated?

toblum, 2019-07-10:
@VeniceNerd The changes from the develop branch were merged to the main branch a bit accidentially. That messed up everything. So you should now have all RGBW changes also in the main project, but they're not complete now and not really tested so far. I think I'll have to do a bit housekeeping to clean up everything.
The mentioned lines can be found in McLighting.ino

DoraDeu, 2020-05-07:
Did anything happen in the meantime in the direction of RGBW integration or housekeeping? 
I think SK6812 RGBW will get more and more important and it would be great if it could be supported.

ChristophA, 2020-05-07:
> Did anything happen in the meantime in the direction of RGBW integration or housekeeping?
> I think SK6812 RGBW will get more and more important and it would be great if it could be supported.

Just use this fork. https://github.com/FabLab-Luenen/McLighting