toblum/McLighting #26: MQTT for WS2812FX working

LeonVos, 2017-07-02:
/*
 MQTT WS2812FX example 
 
  - connects to an MQTT server
  - publishes "hello world" to the topic "outTopic"
  - subscribes to the topic "inTopic"

    On the bottum you will find all the control commands  ;-)
*/

#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <WS2812FX.h>

#define LED_COUNT 26
#define LED_PIN D1
WS2812FX ws2812fx = WS2812FX(LED_COUNT, LED_PIN, NEO_RGB + NEO_KHZ800);

boolean command_complete = false;  // whether the command string is complete

char* topic = "inTopic";  // Change this to the topic where you put the commands.

const char *ssid =  "wifi-SSID";   // cannot be longer than 32 characters!
const char *pass =  "wifi-passw";   // Wifi password

// Update these with values suitable for your network.
//IPAddress server(192, 168, 1, 1);     //example
IPAddress server(192, 168, 192, 53);    // IP address off your MQTT broker

WiFiClient wclient;
PubSubClient client(wclient, server,1883);

#define BUFFER_SIZE 100

void callback(const MQTT::Publish& pub) {
  //Serial.print(pub.topic());
  //Serial.print(" => ");
  Serial.println(pub.payload_string());
  String command = pub.payload_string();
  
 if(command == F("b+")) { 
    ws2812fx.increaseBrightness(25);
    Serial.print(F("Increased brightness by 25 to: "));
    Serial.println(ws2812fx.getBrightness());
  }

  if(command == F("b-")) {
    ws2812fx.decreaseBrightness(25); 
    Serial.print(F("Decreased brightness by 25 to: "));
    Serial.println(ws2812fx.getBrightness());
  }

  if(command.startsWith(F("b "))) { 
    uint8_t b = (uint8_t) command.substring(2, command.length()).toInt();
    ws2812fx.setBrightness(b);
    Serial.print(F("Set brightness to: "));
    Serial.println(ws2812fx.getBrightness());
  }

  if(command == F("s+")) { 
    ws2812fx.increaseSpeed(10);
    Serial.print(F("Increased speed by 10 to: "));
    Serial.println(ws2812fx.getSpeed());
  }

  if(command == F("s-")) {
    ws2812fx.decreaseSpeed(10); 
    Serial.print(F("Decreased speed by 10 to: "));
    Serial.println(ws2812fx.getSpeed());
  }

  if(command.startsWith(F("s "))) {
    uint8_t s = (uint8_t) command.substring(2, command.length()).toInt();
    ws2812fx.setSpeed(s); 
    Serial.print(F("Set speed to: "));
    Serial.println(ws2812fx.getSpeed());
  }

  if(command.startsWith(F("m "))) { 
    uint8_t m = (uint8_t) command.substring(2, command.length()).toInt();
    ws2812fx.setMode(m);
    Serial.print(F("Set mode to: "));
    Serial.print(ws2812fx.getMode());
    Serial.print(" - ");
    Serial.println(ws2812fx.getModeName(ws2812fx.getMode()));
    client.publish("outtopic/modename",ws2812fx.getModeName(ws2812fx.getMode())); // state topic you can use to get the current WS2812FX mode name
  }

  if(command.startsWith(F("c "))) { 
    uint32_t c = (uint32_t) strtol(&command.substring(2, command.length())[0], NULL, 16);
    ws2812fx.setColor(c); 
    Serial.print(F("Set color to: "));
    Serial.print(F("0x"));
    if(ws2812fx.getColor() < 0x100000) { Serial.print(F("0")); }
    if(ws2812fx.getColor() < 0x010000) { Serial.print(F("0")); }
    if(ws2812fx.getColor() < 0x001000) { Serial.print(F("0")); }
    if(ws2812fx.getColor() < 0x000100) { Serial.print(F("0")); }
    if(ws2812fx.getColor() < 0x000010) { Serial.print(F("0")); }
    Serial.println(ws2812fx.getColor(), HEX);
  }
  Serial.println("End of callback");
}

void setup() {
  // Setup console
  Serial.begin(115200);
  delay(10);
  Serial.println();
  Serial.println();
 
  client.set_callback(callback);
  WiFi.status();
  ws2812fx.init();
  ws2812fx.setBrightness(255);
  ws2812fx.setSpeed(175);
  ws2812fx.setColor(0xFFFFFF);
  ws2812fx.setMode(FX_MODE_STATIC);
  ws2812fx.start();

}

void loop() {
     ws2812fx.service();
  if (WiFi.status() != WL_CONNECTED) {
    Serial.print("Connecting to ");
    Serial.print(ssid);
    Serial.println("...");
    WiFi.begin(ssid, pass);
   
    if (WiFi.waitForConnectResult() != WL_CONNECTED)
    {
      return;
    }
      
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
  }

if (WiFi.status() == WL_CONNECTED) {
    if (!client.connected()) {
      Serial.println("Connecting to MQTT server");
      if (client.connect(MQTT::Connect("ws2812fx-xxx")  // change xxx for every ws2812fx client, this is the ID name for each subscriber, can't be the same.
  //     .set_auth("user", "pass"))) {                  // uncomment when you use username and password for you MQTT broker/server.
        Serial.println("Connected to MQTT server");
  client.set_callback(callback);
  client.publish("outtopic","xxx Online");              // Change outtopic to yours, like "home/garden/lights" ,change xxx for CLIENT name
  client.subscribe(topic);                              // this is the incoming topic for control
      } else {
        Serial.println("MQTT error connecting");   
      }
    }

    if (client.connected())
      client.loop();
  }
}

/*
Usage:

m <n>    : select mode <n>, "m 8"  runs Rainbow (without " " ;-)

b+   : increase brightness
b-   : decrease brightness
b <n>    : set brightness to <n>,  "b 255" sets max brightness (without " " ;-)

s+   : increase speed
s-   : decrease speed
s <n>    : set speed to <n>,   "s 255"  sets max speed (without " " ;-)

c 0x007BFF   : set color to 0x007BFF, "c 0xFFFFFF" sets color to white (without " " ;-)

Have a nice day.

0 Static
1 Blink
2 Breath
3 Color Wipe
4 Color Wipe Random
5 Random Color
6 Single Dynamic
7 Multi Dynamic
8 Rainbow
9 Rainbow Cycle
10  Scan
11  Dual Scan
12  Fade
13  Theater Chase
14  Theater Chase Rainbow
15  Running Lights
16  Twinkle
17  Twinkle Random
18  Twinkle Fade
19  Twinkle Fade Random
20  Sparkle
21  Flash Sparkle
22  Hyper Sparkle
23  Strobe
24  Strobe Rainbow
25  Multi Strobe
26  Blink Rainbow
27  Chase White
28  Chase Color
29  Chase Random
30  Chase Rainbow
31  Chase Flash
32  Chase Flash Random
33  Chase Rainbow White
34  Chase Blackout
35  Chase Blackout Rainbow
36  Color Sweep Random
37  Running Color
38  Running Red Blue
39  Running Random
40  Larson Scanner
41  Comet
42  Fireworks
43  Fireworks Random
44  Merry Christmas
45  Fire Flicker
46  Fire Flicker (soft)

*/

marksev1, 2017-07-06:
Could you integrate that into the main sketch so that we have all the features at once? :)

LeonVos, 2017-07-06:
I am working on that. (but time is the problem, be patient)

toblum, 2017-07-08:
Hi @LeonVos,

this is great, I also experimented with that some weeks ago, but did not find any time to dig deeper.

I had a problem to make PubSubClient and WiFiManger work together. This is a known issue: https://github.com/tzapu/WiFiManager/issues/15

I had some progres with https://github.com/btomer/WiFiManager which is a fork of WiFiManger, but this may ba a bad option, as this is not regularly updated.

Regards
Tobias

LeonVos, 2017-07-11:
@toblum, 
MQTT finished !!
Works great, i did not find any issue with the other clients/servers, everything works like a ...... ;-)
I only did not test OTA yet.
When you like it (too) you can ADD this to your MC Lighting (V3 ?) ;-)

Regards Leon Vos

marksev1, 2017-07-11:
So its updated in the first post right? Can we see it on your github too?

LeonVos, 2017-07-11:
Hi, becauce it's a addon on MClighting, i want to keep it here.
I have update the above, just kopie and past it over the main MCLichting sketch page, and make your needed changes on it.
Leon Vos.

toblum, 2017-07-11:
Hi @LeonVos, 
These are great news. I will put your code in a new branch. There we can do some tests before we replace the master branch. 
Regards
Tobias

drschnalli, 2017-07-28:
No matter what i do... i cannot compile the mqtt branch :(

`McLighting.ino: In function 'void setup()':
McLighting:354: error: expected '}' before ';' token
McLighting:354: error: expected ')' before ';' token
McLighting.ino: At global scope:
McLighting:355: error: mixing declarations and function-definitions is forbidden
McLighting:355: error: expected initializer before ')' token
McLighting:357: error: 'server' does not name a type
McLighting:365: error: expected unqualified-id before ')' token
McLighting:369: error: 'server' does not name a type
McLighting:372: error: expected unqualified-id before ')' token
McLighting:374: error: 'server' does not name a type
McLighting:376: error: 'server' does not name a type
McLighting:380: error: expected unqualified-id before ')' token
McLighting:382: error: 'server' does not name a type
McLighting:388: error: expected unqualified-id before ')' token
McLighting:393: error: 'server' does not name a type
McLighting:412: error: expected unqualified-id before ')' token
McLighting:414: error: 'server' does not name a type
McLighting:419: error: expected unqualified-id before ')' token
McLighting:421: error: 'server' does not name a type
McLighting:424: error: expected unqualified-id before ')' token
McLighting:426: error: 'server' does not name a type
McLighting:431: error: expected unqualified-id before ')' token
McLighting:433: error: 'server' does not name a type
McLighting:435: error: expected unqualified-id before ')' token
McLighting:437: error: 'server' does not name a type
McLighting:442: error: expected unqualified-id before ')' token
McLighting:444: error: 'server' does not name a type
McLighting:449: error: expected unqualified-id before ')' token
McLighting:451: error: 'server' does not name a type
McLighting:456: error: expected unqualified-id before ')' token
McLighting:458: error: 'server' does not name a type
McLighting:464: error: expected unqualified-id before ')' token
McLighting:466: error: 'server' does not name a type
McLighting:471: error: expected unqualified-id before ')' token
McLighting:473: error: 'server' does not name a type
McLighting:478: error: expected unqualified-id before ')' token
McLighting:480: error: 'server' does not name a type
McLighting:485: error: expected unqualified-id before ')' token
McLighting:487: error: 'server' does not name a type
McLighting:492: error: expected unqualified-id before ')' token
McLighting:494: error: 'server' does not name a type
McLighting:496: error: expected unqualified-id before ')' token
McLighting:498: error: 'server' does not name a type
McLighting:502: error: expected unqualified-id before ')' token
McLighting:504: error: 'server' does not name a type
McLighting:505: error: expected declaration before '}' token
expected '}' before ';' token`

LeonVos, 2017-07-28:
Hhhmmm i see alot of missing   }}}}}}}} errors , i think you copied wrong.  This week i try to place it on a "new" github, with (and thats importend too) to correct pubsubclient.h.  A link will be put here.

drschnalli, 2017-07-29:
I was a bit confused as i just downloaded the complete zip package of the mqtt branch. I was using Arduino 1.8.x for Windows and i didnt change the libraries with which i could compile the master branch (successfully) before. So i dont know what i'm doing wrong... i think youre using a special PubSubClient or Wifimanager or some other library :(...  i even tried another Arduino IDE (1.6.5) which gave me the same error messages :(

toblum, 2017-08-06:
@All:
I also did not manage to get the above code to work. Maybe problems when I copied the text.

In the meantime I got my initial MQTT implementation also running. I created a new branch "mclighting_mqtt_pubsub". My implementation uses the same codes as the websockets protocol, so no changes here.
The MQTT credentials can be configured via the WiFiManager.

Maybe you all can have a look at it and maybe it's a good idea to merge it with LeonVos code. What do you think @LeonVos ?

toblum, 2017-08-19:
I will close this issue now just to cleanup a bit.

Tobias

closed 2017-08-19 by toblum