toblum/McLighting #29: MCLighting in "accesspoint"

Xav83130, 2017-08-04:
Hello,
I have a small question, is it possible to use MCLighting without going through the wifi of my network? (In access point)
If so, how?
thank you in advance.

And congratulations for your code and thank you for sharing.

LeonVos, 2017-08-04:
And why would you want that ? ☆for if you don't have local wifi, maybe use your phone as A.P. , and then the mclighting and your phone have the same range, and the mclighting server will run. ●i did not tried this, but i think it will work

toblum, 2017-08-05:
@Xav83130 This is not possible, because McLighting wasn't built this way and would need serious changes to make that work. Using your phone as a AP might be a good workaround as @LeonVos suggested.

closed 2017-08-05 by toblum

Xav83130, 2017-08-05:
OK thanks a lot