toblum/McLighting #30: Hotspot server mood

Fireflaker, 2017-08-08:
How can I modify the code to make it work as a wifi server instead of a client?

Fireflaker, 2017-08-09:
`*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 4

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: cont 
sp: 3fff2f20 end: 3fff33f0 offset: 01a0

>>>stack>>>
3fff30c0:  4024fdf9 00000017 60000200 40107508  
3fff30d0:  40228129 4024c76a 00000005 40221056  
3fff30e0:  000100c4 4022361f 3fff564c 3fff5264  
3fff30f0:  4022836a 3fff564c 3fff5264 3fff2110  
3fff3100:  00000000 3ffef710 3ffef68c 3fff5264  
3fff3110:  3fff2110 3fff2110 00000001 3fff2110  
3fff3120:  00000018 00000064 06a620a2 fffeffff  
3fff3130:  3ff20a00 0000ffff 3fff3158 40223cc2  
3fff3140:  3ffef65c 3fff5264 3fff5264 3fff2110  
3fff3150:  06a620a2 3fff3ada 00000000 00000000  
3fff3160:  00000000 00000000 00000000 00000000  
3fff3170:  00000000 00000000 00000000 00000000  
3fff3180:  00000000 00000000 00000000 40224e14  
3fff3190:  3fff5264 00000001 00000000 40224e5c  
3fff31a0:  00000004 3ffef710 3fff0468 00000000  
3fff31b0:  40234801 00000003 00000003 000003fd  
3fff31c0:  40234951 00000003 00000001 3fff2110  
3fff31d0:  3fff4ddc 402349c6 00000003 3ffe9aac  
3fff31e0:  4020b181 00000000 00000000 40217d18  
3fff31f0:  3ffe92f8 00000000 3fff3280 4021078b  
3fff3200:  3fff22dc 00000238 00000238 4010020c  
3fff3210:  00000004 3ffe9aac 3fff324c 4010068c  
3fff3220:  3ffe92f8 4025595c 3fff2304 00000000  
3fff3230:  00000004 3ffe9aac 3fff3280 40210911  
3fff3240:  00000000 00000000 00000000 00000000  
3fff3250:  00000000 00000000 3fff2304 4020f03a  
3fff3260:  3fff217c 3fff332c 3fff3280 3fff2304  
3fff3270:  3ffe9aac 3fff2088 3fff20b0 40209ba6  
3fff3280:  00000000 00000000 3ffe9edc 00000000  
3fff3290:  3fff4dac 0000000f 00000000 3fff4dc4  
3fff32a0:  0000000f 00000000 00000000 00000000  
3fff32b0:  00000000 3ffea700 00000000 3ffea700  
3fff32c0:  00000000 3ffea700 00000000 3ffea700  
3fff32d0:  00000000 3ffea700 00000000 3ffea700  
3fff32e0:  00000000 00000004 ffffffff fe000001  
3fff32f0:  3ffe92f8 00000000 fe01ef35 40207958  
3fff3300:  40206ddc 3fff3368 3fff3354 3fff3340  
3fff3310:  3fff332c feefeffe feefeffe feefeffe  
3fff3320:  feefeffe feefeffe feefeffe 3ffe9520  
3fff3330:  3ffe9528 3fff4d84 00000020 3ffe92f8  
3fff3340:  3ffe950c 3ffe9514 3fff4d5c 00000020  
3fff3350:  3ffe92f8 3ffe94f8 3ffe9500 3fff4d4c  
3fff3360:  00000006 3ffe92f8 3ffe94e0 3ffe94e8  
3fff3370:  3fff4d04 00000040 3ffe92f8 feefeffe  
3fff3380:  feefeffe feefeffe feefeffe feefeffe  
3fff3390:  feefeffe feefeffe feefeffe feefeffe  
3fff33a0:  feefeffe feefeffe 3fff4cec 0000000f  
3fff33b0:  00000001 feefeffe feefeffe feefeffe  
3fff33c0:  feefeffe feefeffe feefeffe 3fff23c0  
[... 84 lines cut ...]
3fff33e0:  feefeffe feefeffe 3fff23d0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
⸮*WM: Adding parameter
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster`

This repeats forever. Any fix please?

thundergreen, 2017-08-10:
I am facing exactly the same issue here :( 

```
*WM: host
*WM: Adding parameter
*WM: port
*WM: Adding parameter
*WM: user
*WM: Adding parameter
*WM: pass
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
```

LeonVos, 2017-08-10:
Hhmm are you using different wifimanager, maybe you need to update or try another ??

toblum, 2017-08-10:
Hi,
I could not reproduce this here. Will look into it soon. Have you enabled MQTT? Does it occur when you disable it?
You could also try the latest version before changes: https://github.com/toblum/McLighting/tree/5f7733ebd335d55610cbe0bfb6ad8fcd1168bda0

Regards 
Tobias

toblum, 2017-08-10:
,rst cause:2, boot mode:(3,6) could also be a Problem with the power supply. How do. you power your ESP?

thundergreen, 2017-08-10:
```
Connect 192.168.1.22 1883
Open http://192.168.1.59/ to open McLighting.
Use http://mc_lighting.local/ when you have Bonjour installed.
New users: Open http://192.168.1.59/upload to upload the webpages first.

FS File: /index.htm, size: 13.52KB
FS File: /edit.htm.gz, size: 4.02KB
FS File: /offline.appcache, size: 385B

Attempting MQTT connection... connected!
MQTT topic in: mc_lighting/in
MQTT topic out: mc_lighting/out
WS: [0] Disconnected!
WS: [0] Disconnected!
WS: [0] Disconnected!
WS: [0] Disconnected!
WS: [0] Disconnected!
WS: [0] Disconnected!
handleFileRead: /
handleFileRead: /offline.appcache
handleFileRead: /
handleFileRead: /offline.appcache
WS: [0] Connected from 192.168.1.207 url: /
handleFileRead: /edit.htm
WS: [0] Disconnected!
handleFileRead: /index.htm
handleFileList: /
WS: [0] Disconnected!
handleFileRead: /index.htm
WS: [0] Disconnected!
handleFileRead: /wlan_reset
WS: [0] Disconnected!
handleFileRead: /edit.htm
handleFileRead: /index.htm
handleFileList: /
WS: [0] Disconnected!
WS: [0] Disconnected!
handleFileUpload Name: /graphs.js.gz
handleFileUpload Size: 1971
handleFileList: /
handleFileRead: /graphs
WS: [0] Disconnected!
WS: [0] Disconnected!
WS: [0] Disconnected!
handleFileRead: /
handleFileRead: /offline.appcache
handleFileRead: /
handleFileRead: /offline.appcache
WS: [0] Connected from 192.168.1.207 url: /
WS: [0] get Text: #808080
Set main color to: [128] [128] [128]
WS: [1] Disconnected!
```

thundergreen, 2017-08-10:
what is strange that i can connect to the esp but dunno how it is working.. it is not connected to my router. And I dunno how to make it connecting to this /wlan_reset won't work I used exactly the same wifi manager

thundergreen, 2017-08-10:
[image]
i use this wifi manager and yes mqtt is enabled.. it sonnected to the mqtt server is u can see above but then disconnects

thundergreen, 2017-08-10:
Can I somehow set the routers SSID and password to connect diectly?
I dunno how it it connecting as i NEVER entered my Router WIFI credentials

thundergreen, 2017-08-10:
has someone here implmented this in home assistant ? I just dont get it running no matter what i do

thundergreen, 2017-08-11:
Unfortunatelz i could not get this running on my ESP. I finally gave up It
connects temporairlz but disconnects right after that  no idea how to
proceed

Fireflaker, 2017-08-17:
I got it fixed by factory resetting my ESP8266 and re-flash the old version firmware. Later versions do not work

Fireflaker, 2017-08-17:
[binFIRMWARE.zip](https://github.com/toblum/McLighting/files/1232347/binFIRMWARE.zip)

please use ESP_8266_BIN0.92.bin

Fireflaker, 2017-08-17:
[ESP8266Flasher.zip](https://github.com/toblum/McLighting/files/1232352/ESP8266Flasher.zip)

thundergreen, 2017-08-18:
how would that be done. I mean the steps. is that bin file all i need to
copy on the esp?

toblum, 2017-08-19:
Are there any problems still open?

If you want to use the old version without MQTT, you could always get it at: https://github.com/toblum/McLighting/tree/5f7733ebd335d55610cbe0bfb6ad8fcd1168bda0
(Disabling MQTT in the latest version should lead to the same result)

noisufnoc, 2017-08-22:
i'm having the panic/crash issue here with MQTT enabled.  I disabled in definitions.h and it works normally.

noisufnoc, 2017-08-22:
Okay, I think I figured out the MQTT crashing issue.  I noticed that both @thundergreen and I use home-assistant.  More than likely we are using PubSubClient for other things in our enviornment, and in my case I'm using Sonoff-Tasmota which uses a larger MQTT_MAX_PACKET_SIZE 

https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisite#installing-external-libraries

I changed my PubSubClient.h back to default and I was able to re-compile and boot my ESP

toblum, 2017-09-29:
@noisufnoc I also noticed issues regarding this setting. Maybe this will be fixed in PubSubClient in a later release. In the meanwhile we have to adjust the setting according to the project.

closed 2017-09-29 by toblum

thundergreen, 2017-10-02:
Can someone please point me to the right solution here? It seems a bit confusing for me which PubSubClient to akte and how to configure now

toblum, 2017-10-03:
@thundergreen If you use the latest PubSubClient without modifications it should work without problems. If you patched the PubSubClient for some reason (Tasmota or somethin glike that), you may have to reverse that change temporarily.