toblum/McLighting #32: Memory leak in MQTT callback function

nimbl, 2017-08-11:
https://github.com/toblum/McLighting/blob/4820d584422f595cf78ceb40c38ea5aab443f503/Arduino/McLighting/request_handlers.h#L310

The use of "malloc" without "free"ing up the result causes a memory leak - free heap space drops steadily with each received mqqt message, eventually resulting in heap overflow and chip reset. Suggest avoiding the use of malloc entirely. Just use "payload_in" instead of defining and copying the contents.

toblum, 2017-08-11:
You're right. I missed that. Will have a look at it as soon I find time. 
Thank you!

toblum, 2017-08-19:
Dear @nimbl,
I tried using payload_in directly but had problems, so I ended up using free().
--> https://github.com/toblum/McLighting/commit/ab9253abde7350ba504fa5a21a38ee7aa7ccc5c2
Thanks again!
Regards
Tobias

closed 2017-08-19 by toblum