toblum/McLighting #33: Random Crashes - Red Pulsing LED Strip

drschnalli, 2017-08-16:
Hello, i found out, that after a while (maybe 1 hour or more) something strange happens. The Strip turns into breathe mode with color red (as if he wants to say something is wrong). It happens when its in off mode and also when its running one of the effect modes. When i plug the esp8266 into the pc and leave it running i cannot reproduce the error. On my mqtt Server i can see serveral reconnects (approx. every hour). 

Does someone know what is causing that? Maybe i try to disable MQTT (but with my mqtt connection everything is okay, even the controls via mqtt are working). 

When i use the better-http-rgb plugin within homebridge i also face the issue that my devices are not responding when the strip starts breathing red, because the webserver is also not responding then. When i reboot the esp8266 manually then, everything is fine again until the next crash.

toblum, 2017-08-19:
Dear @drschnalli,
there was a problem with MQTT that is fixed now (I hope), but also before I could not reproduce this. When you leave I plugged into the PC, is there a error message in serial console? 
I had a similar problem some time ago (reboots after a varying timespan), but that was related to the power supply.
Regards
Tobias

toblum, 2017-09-20:
Hallo @drschnalli,

wenn der Strip neu startet, dann steht der Grund in der Console. Meist geht es dabei um die Stromversorgung. Blöd, dass du bei angeschlossener Console den Fehler nicht hast.

Du könntest mal testen, ob er auch auftritt, wenn du ein anderen Skript mit Effekten länger laufen lässt, z.B. https://github.com/kitesurfer1404/WS2812FX/blob/master/examples/auto_mode_cycle/auto_mode_cycle.ino dann wüsste man, wo der Fehler zu suchen ist.

Der MQTT Fix ist im letzten Commit im master branch bereits drin.

Viele Grüße
Tobias

drschnalli, 2017-10-25:
Ich hab es mit den neuesten Files getestet und gestern Abend denselben Fehler festgestellt. Diesmal hatte ich sogar eine statische Farbe ausgewählt und nach ca 1,5 Stunden wechselte er dann zurück auf das default preset. Danach hab ich geschaut ob er im Off zustand auch wieder abstürzt und dies war dann nach ca 30 minuten der Fall. Ich werde mal schauen ob das Wechseln der Stromversorgung hilft, einen weiteren NodeMCU hätte ich auch noch herumfliegen. Derzeit verwende ich ein China 5 Volt 2 Ampere Steckdosennetzteil. Ich werde es mal durch ein Stabiles PC Netzteil ersetzen und es so versorgen oder es mal mit einem Kondensator an der Stromversorgung es MCU versuchen.

toblum, 2017-10-28:
Hallo @drschnalli,

kannst du mal mit http://McLighting_hostname/esp_status schauen, ob der freie Speicher mit der Zeit weniger wird, wenn das Problem auftritt? Nur um evtl. Speicherprobleme ausschließen zu können.

Viele Grüße

closed 2018-01-13 by toblum