toblum/McLighting #35: New Functions

Snezhana92, 2017-08-25:
Create drop-down list with effects instead of buttons. It will be better. Also selection of favourite effects by the tick and they could work by turns.

toblum, 2017-08-26:
Dear @Snezhana92,

the buttons were chosen to make usage especially with touch devices easier. You need only one click to use a button but at minimal two for drop-downs. But you could easily patch the HTML-file to make it work your preferred way.
Favourite effects could be a good idea, I will note that.
What do you mean with "hey could work by turns"?

Regards
Tobias

Snezhana92, 2017-08-29:
when there will be updates?

toblum, 2017-08-30:
As soon as I (or another contributor) finds time do implement something. There are no regular releases.

Vika1283, 2017-09-07:
> 
The drop-down list saves space on the screen and looks great.

--------------------------------------------------
What would you mean if "hey can work in turn"?
Select the effects that you liked with checkmarks and click the "Start" button

Snezhana92, 2017-09-07:
The drop-down list saves space on the screen and looks great.

--------------------------------------------------
What would you mean if "hey can work in turn"?
Select the effects that you liked with checkmarks and click the "Start" button

toblum, 2017-09-29:
Thanks to another user who contributed code there is now a possibility to auto change effects.

You could also try this project that is derived from McLighting: https://github.com/doctormord/Responsive_LED_Control

closed 2017-10-28 by toblum