toblum/McLighting #37: Couple of observations and a request

rhamblen, 2017-09-10:
Firstly thanks!  I'm planning halloween and christmas lights early! and wanted to have a nice setup.
I've got two rolls of WS2812 lights and have been testing compared to the standard controller.

I much prefer some of the light patterns from your collection but also I like that I can use the mobile device to set things up,
Its actually got me thinking about upgrading several existing light strings (currently WW only) and getting a super effect for the room.  I digress!

Observations:  
When I have static colour I notice a flicker.  The flicker is random and when the light is dim. Also on a couple of sequences I noticed a random (as in not correct) sequence occuring occasionally.  I am assuming this is a problem with noise on the data signal as opposed to the software, but wanted to flag it in case you are aware of a problem.

Request:
Some controllers you can purchase have the ability to save a number of the sequences to create your custom show.  ( see DIY settings in [led2013-x](http://www.g-nor.com/upload/2013111509064823.pdf) ) 
Would be good to have this capability.

The other thing I would like to do is create a customer profile which I dont mind doing at compile time.
Basically rather than having the whole string static, I want to define the areas to turn on.  This is because the string will run along the front of the house on the soffit board (roof line) and I want the light to shine down to the sides of the window and light the shutters not the windows or wall.  
A simple string of ones and zeros representing each pixel would be easy to create.  This could then be added to the list of profiles and you control colour and brightness as per normal.

Thanks

toblum, 2017-09-20:
Dear @rhamblen,

thank you for your feedback i'd love to see your setup in action some day.
I never noticed such a flicker, but I don't own very long strips at the moment. Could be noise on the data line. Also connecting the Ground on both sides of the strip helped me sometimes.

There is a pull request from another user that provides support for own auto-cycle schemes. I created a new branch for that to test it out: https://github.com/toblum/McLighting/tree/feature/mclighting_autocycle

I noted your enhancement proposal on the list in the readme.

Regards
Tobias

toblum, 2017-09-29:
This functionality is now in the master branch.

closed 2017-11-25 by toblum