toblum/McLighting #38: Bug in listStatusJSON() function

moose4lord, 2017-09-14:
Hi, I've started playing with McLighting, and like it very much. It's going to make my Halloween project much easier. Thank you for sharing your hard work.

But I did find a bug I wanted to bring to your attention. If I knew GitHub a little better, I'd fix it and submit a pull request, but I'm not that familiar with GitHub.

Anyway, if I run McLighting and browse to http://{ESPHOSTNAME}/status, my ESP crashes. I peeked at the code and in request_handlers.h there's a function called listStatusJSON(), which is supposed to return a collection of status parameters in JSON format. The function uses snprintf() to compose the JSON string, which uses the WS2812FX function getModeName() to retrive themode name. Unfortunately WS2812FX stores the mode names in FLASH (i.e. PROGMEM) memory and snprintf() expects the value returned by getModeName() to be in RAM. More about PROGMEM here: [http://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html]

I added a couple lines to listStatusJSON(), which copies the WS2812FX mode name from FLASH to RAM and uses the copy in snprintf(). That fixed the problem for me. Here's my version of listStatusJSON():
`char* listStatusJSON() {
  char json[255];

  char modeName[30];
  strncpy_P(modeName, (PGM_P)strip.getModeName(strip.getMode()), sizeof(modeName)); // copy from progmem

  snprintf(json, sizeof(json), "{\"mode\":%d, \"ws2812fx_mode\":%d, \"ws2812fx_mode_name\":\"%s\", \"speed\":%d, \"brightness\":%d, \"color\":[%d, %d, %d]}",
    mode, strip.getMode(), modeName, ws2812fx_speed, brightness, main_color.red, main_color.green, main_color.blue);
  return json;
}`

Hope that helps. If I'm not being clear, please let me know.

moose4lord, 2017-09-14:
Hmmm, I looked at the latest WS2812FX library and the change to conserve memory by moving static strings from RAM to FLASH seems to have been made recently (2 months ago). So this problem of ESP crashing only happens when using the latest version of WS2812FX. Just FYI.

toblum, 2017-09-20:
Hi @moose4lord,

thank you very much. I will have a look at it as soon as I find time to try everything out.

Regards
Tobias

toblum, 2017-09-20:
@moose4lord,

did I see correctly that this fix was also contained in your pull request with auto cycle?

toblum, 2017-09-29:
I just merged your pull request into the master branch. Thank you for contributing!

Regards
Tobias

closed 2017-09-29 by toblum