toblum/McLighting #41: Request: NTP Function

Malle355, 2017-09-22:
Hi,

is it possible to add an NTP (Network Time Protocoll) Function? So that my lights turn on in the evening and turn off in the morning? So you can set the time in the Webinterface.

That would be really awesome.

Thanks.

toblum, 2017-09-22:
Hi @Malle355 ,

I think this might be possible from the technical side, but would be a big effort. I think this is something that could easily solved by a automation software via MQTT or websocket/REST request. I use NodeRed for such things, but other solutions should work also.

Regards
Tobias

closed 2017-09-22 by toblum

Malle355, 2017-09-22:
Danke. Ich werde mal schauen ob ich eine Möglichkeit finde.

Malle355, 2017-09-27:
Hello,

I just got MQTT working now and its perfectly working with Node-RED. Is it possible to get all the Modes working with MQTT? There are only the Modes which can be requested via Websocket but there are many more on the Mode Site.

[URL=http://www.directupload.net][IMG]http://fs5.directupload.net/images/170927/ah2zsyq3.png[/IMG][/URL]

[URL=http://www.directupload.net][IMG]http://fs5.directupload.net/images/170927/98aeysy8.png[/IMG][/URL]

toblum, 2017-09-29:
Just have a look here: https://github.com/toblum/McLighting/wiki/WebSocket-API
All Websocket commands (except ~ work). Your could call each animation by /<animation id>, e.g. "/8"