toblum/McLighting #42: Request: More Modes via MQTT

Malle355, 2017-09-28:
Hello,

I just got MQTT working now and its perfectly working with Node-RED. Is it possible to get all the Modes working with MQTT? There are only the Modes which can be requested via Websocket but there are many more on the Mode Site.

[image]

[image]

Thank you.

toblum, 2017-09-29:
Just have a look here: https://github.com/toblum/McLighting/wiki/WebSocket-API
All Websocket commands (except ~ work). Your could call each animation by /<animation id>, e.g. "/8"

closed 2017-09-29 by toblum

Malle355, 2017-09-29:
Ah thanks. Sorry I didn't see that. Thank you that much.