toblum/McLighting #43: doesn't work properly, static blinks

sladaar, 2017-09-29:
Hello! And thanks for listening me and i hope i did everything correctly.
I've managed to work it all, but i have this issue especially visible in static mode: the leds are lighting up like in "color wipe random" mode or similar, but they don't go static.
I have three 5m ws2812b connected to the same power supply and the data wire comes from the nodemcu, then it's connected in series from the first to the third strip.
The "scan" mode work properly i think.

Thanks in advice and sorry for my english, i'll try my best!

toblum, 2017-09-29:
Hi @sladaar 
I never tried auch an amount of LEDs. Could be an issue with signal quality maybe. Does it work when you connect only 1m of LEDs?

LeonVos, 2017-09-29:
Thats 15 m !! 
How BIG  is your (stable) power supply ? 
I think you need about 40 Amp !
Or try to power each strip at the begin. All that power can't flow to the end off that 15 m.
Does it twinkle to when you dim to say 25% ?

sladaar, 2017-09-29:
many thanks for the reply to both of you!

@toblum yes i tried, and the problem seems to be less dramatic! I mean, i tried with 5m only (i didn't want to cut the roll at the moment to try with 1m) and half strip seems static (i say "seems" because the colour it's not really static and solid, it slightly fades), the other half blinks and changes colour clearly.
I thought a signal loss too, but i don't know how to "stabilize" it!

@LeonVos Yeah i know it's a large amout of leds, i'm trying to put them on the ceiling!
The PS is rated 60A, i bought a bigger one to be sure it could manage all the strip.
I connected every strip (5m) to the PS, and only the data pin in series between each strip (5m).
Do you thinks it's better to "give" power even at the end of the 5m?
And sadly yes, it twinkles even with low brightness (i also thought that too muc current was hard to handle).

Ah, i even tried with a capacitor on the power line!

Thanks again, i really feel it must be a signal problem but i don't figure out how to protect it (i even tried with ferrite around the data cable...)

toblum, 2017-09-30:
@sladaar Unfortunately I don'thave experience with big strips, but it does look like a signal or power issue. Also powering from the end might help, but you could have a look at the really informative Adafruide Neopixel Uber-Guide: https://learn.adafruit.com/adafruit-neopixel-uberguide/basic-connections

sladaar, 2017-10-02:
@toblum i'll try as soon as i can, thanks again for the help!

closed 2018-01-13 by toblum