toblum/McLighting #44: handleSetSingleLED not working correctly

yanbec, 2017-10-02:
Hi!
At first, many thanks, I like this project very much :)
While enhancing it a bit by open the possibility to manually set different colors to several specific LEDs at once via WebSockets, I think I found a bug in the handleSetSingleLED()-function.
I currently have small testing matrix connected to a NodeMCU with 3*12 LEDs.
Steps to reproduce: 
(1) Send "!16aa6f46" via WS.
Reaction on serial monitor:
> WS: [0] Connected from 192.168.1.169 url: /
> WS: [0] get Text: !16aa6f46
> WS: Set single led [22] to [170] [111] [70]!

Alternative testing snippet in JS:

```
var led = 0;
connection.onopen = function () {
  console.log("Connection opened");
  for (var i = 0; i < 37; i++) {
	setTimeout(function() {	
		var stringToSend = "!" + led + "0000FF";
		console.log(stringToSend);
		connection.send(stringToSend);
		led++;
	}, 50*i); //delay to avoid crashing the ESP because of too many requests
                    //which is the initial reason for the new command
	
}
};
```

With the snippet above only the following LEDs turn on: 1-10;  17-26; 33-36

I already tried to determine what exactly is causing this "conversion" from one Led-number to another, but wasn't able to. I don't get how this function works exactly ;)

Best regards,

toblum, 2017-10-03:
I thought I had this fixed some time ago. I will have a look at it. Also thank you for your contribution.

toblum, 2017-10-28:
Hi @yanbec,

thanks again for your bug report. I fixed this. The number was interpreted as hex instead of a decimal number.

Regards
Tobias

closed 2017-10-28 by toblum