toblum/McLighting #46: void configModeCallback (WiFiManager *myWiFiManager) {

Desacrel, 2017-10-08:
Hi, n1 Idea. 
But I cant Test it, 

McLighting:117: error: variable or field 'configModeCallback' declared void

 void configModeCallback (WiFiManager *myWiFiManager) {

                         ^

McLighting:117: error: 'WiFiManager' was not declared in this scope

McLighting:117: error: 'myWiFiManager' was not declared in this scope

yanbec, 2017-10-08:
Hi!
You probably didn't install the WiFiManager-lib, which is required for easily configure the connection.
-> https://github.com/tzapu/WiFiManager (can be installed via Arduino IDE).

Regards,

Desacrel, 2017-10-08:
OMG.
I Have it over 10 times checked 
but show everytime 
#define WiFiManager_h   
but not #include <WiFiManager.h>
... sry can i Kill this Post ? ^^

Now i test it

Sry for Bad EN. Im from Ger

Lg Desa

yanbec, 2017-10-08:
You're welcome ;)
I think @toblum will close the issue when he has the time.

Desacrel, 2017-10-08:
But I have Trouble with the manager.
Take i Manual the SSID and PW With " WiFi.begin(ssid, password); "
They go Online.
but they dont go in AP Mode without this. 
The led at Wemos blink 4 Times and then off, at MCU ist 3 times blinking and then all 5 sec.

They dont give me a wlan ap
Take i manual the WIfi. Port 81 ist only for Sockets 
and Port 80 ist dead No Files

yanbec, 2017-10-09:
Did follow the [last step of installation](https://github.com/toblum/McLighting/wiki/Software-installation#upload-the-webpage-to-the-spiffs-filesystem) in the wiki?

The WifiManager only creates an AP if no connection is possible. I don't know if it is possible to choose to create an AP. Anyway, "Port 80 dead" sounds just like you didn't upload the files. 

Another thing I experienced is the ESP storing the WiFi information through reflashing - I had my NodeMCU I use for this project in another one before, so I never saw the WiFiManager because it already had a valid connection.

Desacrel, 2017-10-09:
Oki i test it later

Desacrel, 2017-10-09:
Hm The WifiManager dont work, They dont change to AP Mode. I Test it now withou The Manager and take The Original Client Mode. But i Thinkt the Manager is the better Idea vor other Persons

toblum, 2017-10-20:
Hi @Desacrel,
could you make it work in the meanwhile?
Regards
Tobias

Desacrel, 2017-10-20:
Yes and No, I give him the SSID and pw. The Manager dont work for me.

Your Version work, but make everytime restarts. I search the Problem but im not sure i have the right Power vor Ring ( i think to low Ampere ) 
My time is at these time low, I have 3 Other Projekts on my letter.
I think im testing at next Weekend more. When all Ok i add test fastled Lib and test it together

But Thanks for answer

closed 2017-11-25 by toblum