toblum/McLighting #47: Random reboots

yanbec, 2017-10-16:
Hey folks!
I'm still experimenting with this project, currently I have the little LED matrix I mentioned before, plus a strip of 128 LEDs. Each of them are driven with a NodeMCU-Board with a sufficient power supply.
I wanted to drive them with a Node-RED server, which I deactivated for now, because I thought the constant WebSocket connection (MQTT will come) could be the problem.
What's the problem? If I turn the LEDs off ( -> ws: =off), after a more or less long while, the lights begin to use the "default" animation (rainbow cycle in my case). At the same time the WebSocket connection gets lost, so I assume the ESP just reboots. It does not seem to happen when I use a constant color, but I am not totally sure of that.
When I turn them off for the night, I can be >90% sure they are on when I wake up ;)

I will set the default to "off" for now, then I can maybe see if the problem persists or this works as a workaround for me. Does anybody else have such problems?

Thanks and best regards,
Yannick

toblum, 2017-10-20:
HI @yanbec,

those problems were reported in the past, but were mostly related to bad or aat least not perfect power supply. The ESP is quite sensitive here.
I also had problems with a specific NodeMCU board that were gone after replacing it.
It would be good to know what the ESP prints out to serial when the restart occurs.

Regards
Tobias

yanbec, 2017-10-20:
Hey Tobias,
thanks for your answer.
I worked around by setting the default mode to "off". The problem does not seem to occur when actually using the lights. Probably the PSU is not that stable when nearly at 0% utilization. If that's the case, maybe an additional capacitor for the ESP could solve the problem. I'll check if I find the possibility to "long-time" monitor the serial output and get something from there. 

Regards,
Yannick

cosmocracy, 2017-10-25:
+1, I also have this issue with my LinkNode D1 (ESP8266 board) and my LED strip.  As @yanbec reports it seems to only happen after running lighting effects for a while, not when a constant color is displayed.  

Is there already an issue for persistently storing the current mode so that if/when the ESP8266 reboots (whether a software issue or power issue) it will just pick up with the same color/effect (or OFF state)?

Great project, @toblum, thank you!

toblum, 2017-10-25:
@cosmocracy 
I will have a look into this, bt it's hard to spot because it can't easily be reproduced. If anyone has an idea what causes this, this would be helpful.

I added an todo: https://github.com/toblum/McLighting/blob/master/README.md#todos
If anyone wants to implement this ...

drschnalli, 2017-10-25:
I'm experiencing the same, when im running effects for a long time it will reboot and switch to the default preset (stored in the config file). I will try to attach a capacitor between gnd and vin. Or try to power it via my old PSU instead of a 2A 5v power source.

yanbec, 2017-10-25:
> If anyone wants to implement this ...

Maybe I'll find time on the weekend or the beginning of the next week. Sounds like a nice feature even if the "reboot problem" gets sorted out.

> I will try to attach a capacitor between gnd and vin

That would be nice, maybe the ESP is just a little too sensitive sometimes. Regarding the PSU: I had the problem with 128 LEDs on a 5V 10A supply. The used strip should need about 18W/60 LEDs in full brightness white. That's why I said my supply is "sufficient" ;) Despite that I can reproduce the problem only in "off" mode. 

For me the solution with the default "off" mode still works perfectly in combination with an echo dot & node red.

toblum, 2017-10-28:
Hi @all,

it could also be some kind of memory leak, especially when you use longer strips. I checked, but my longest strip is 60 LEDs. 

Maybe someone who has this problem can check the free memory with http://<McLIghting>/esp_status and see if it's getting less with the time. Here I have about 20.000 to 24.000 bytes free the most time.

What modules do you have enabled? OTA and/or MQTT?

Regards
Tobias

top44, 2017-11-07:
hello,

i have always had problems with constant reboots. Then I tried this [here](https://blog.thesen.eu/esp8266-reset-probleme-loesen-und-relais-stabil-schalten/), since then it works. I use 300 WS2812b leds.

Regards
Holger

cosmocracy, 2017-11-12:
@toblum, I've switched using the MQTT API so even if/when the ESP8266 reboots it will restart the last command sent (as long as I use the "retained" flag for the message).  Is it possible to stack together multiple directives in the API?  That way I could publish a retained message like `*00f2f2;/41` or something to get both the color (cyan) and the effect (fire).  Just a thought.  Thanks!

toblum, 2017-11-12:
@cosmocracy Using retain is a great idea I didn't think of yet. Thanks for sharing that.

Combined commands are not possible yet, but I'll keep that in mind.

dunk8888, 2017-11-16:
The reboot issue and boot loop problem is corrupt flash,the esp sometimes comes with cheap chinese flash chips,your better replacing the chip with winbond chips,this has fixed the issue with all my esp modules,it can be a power thing but its most likely the flash,adding this line WiFi.persistent(false); at the start of your sketch can help too,your esp saves the wifi password and ssid to a spot on flash that gets corrupt,and on booting it looks for this and if corrupt will reset over and over thats the boot loop problem. I bought some esp-12 modules and worked fine till ide flashed them around 20 times then this issue started,the flash chips seem to have a short life,so winbond is the way to go,hope this helps.cheers

toblum, 2017-11-18:
@dunk8888 Yes, I have also suspected that often. I have ESP boards that gave up working after only a few flashing cycles.

Someone wrote that he had success with the tips in this article:
https://blog.thesen.eu/esp8266-reset-probleme-loesen-und-relais-stabil-schalten/ (german)

http://translate.google.com/translate?js=n&sl=auto&tl=en&u=https://blog.thesen.eu/esp8266-reset-probleme-loesen-und-relais-stabil-schalten/ (translated)

Oehoeboeroe, 2017-12-14:
Solution might be to add "#define BUILTIN_LED 2" in the definitions.h
For me this solved a lot of 'strange behavior. 
/BR

Malle355, 2017-12-18:
I also got random reboots. Tried different things like a new power supply or adding a capacitor in front of the esp. Any Ideas?

LeonVos, 2017-12-28:
Hi, i had the same problem, i have 8 mclighting "lights" in my back garden and a WAP in the barn, so 8 wifi connection always from the esp's . the 2 most far from that WAP in the barn did lose the wifi from time to time, and went back to default after connecting again (so , it looks like a reboot, but is a reconnonection)
Now a have those two connected on my house wifi and work perfect like the others.
So maybe you can check if there are to many or hikups in you wifi ?? (just to make sure)

toblum, 2017-12-28:
Hi @LeonVos and @Malle355,

there is a new branch available with a new version that saves the last state after restart. In my tests it worked good. Maybe you want to try it out too. If everything is OK, I will publish it for all.

Regards
Tobias

LeonVos, 2017-12-29:
@toblum, that i will do, good work.
But will be in the NEW year.
SET ALL YOUR MCLICHTING'S ON "RANDOM FIREWORK" and have a great newyears night.
;-)

toblum, 2018-01-20:
I will close this issue for now. Feel free to write a message if there is anything.

closed 2018-01-20 by toblum