toblum/McLighting #48: 2 operation modes: McLighting + Other

dimiHH, 2017-10-20:
Hi,
first of all, thank you for the great work. A very useful and easy to use library. After installing it I decited to build a desk lamp based on it.

My desk lamp should have two modes: Funky (i.e. McLighting) and regular (i.e. white light, dimable with a poti). I want to switch between these two modes with a regular button/switch.

My initial idea was: Include a while loop in the somewhere inside the main lood which holds the very small dimable-white-light-code. while the button is in  "regular-mode" only the code inside the while loop should be repeated. When I switch to "McLighting-Mode" everything _except_ the code in the while-loop should run.
Like this: 

```
while(digitalRead(buttonPin)==HIGH){
    // code for regular mode (poti --> dimable white lights
 
  }
```

But when I put this code in the main loop nothing works anymore. 

Is there a place I can put this while-loop so it works as I want to?
Is it possible to use the analog-pin (A0)? It is used as an imput in the main sketch, but I don't know what for. (and what about the other pins? (at least one is needed as a "buttonPin".

Thanks!
Dimi

toblum, 2017-10-20:
Hi @dimiHH,

thanks for the good feedback.

Regarding your idea: Wouldn't it work to check for button presses in the main loop() and change the mode according to your keypresses. So you could cycle through predefined modes.

Analog A0 should be used. Where did you see that? You're free to use the pins that are not covered be the LED strip.

Regards
Tobias

dimiHH, 2017-10-21:
thanks for the quick response @toblum , your approache also seems more elegant (it makes it  possible to switch to to other predifined modes via the the browser without "unswitching" a physical button first)

So as  see it, the following things need to be done for the changes I want:

1) define a function "whitelight" somewhere in colormodes.h
```
void whitelight () {
         
	checkForRequests();
	if (exit_func) {
		exit_func = false;
		return;
	} 
         // is the upper part necessary?:

    strip.setColor(0,0,0);
    strip.setMode(FX_MODE_STATIC);
    strip.setBrightness(analogRead (map(0, 1024, 0, 255) );
}
```

2) Add the whitelight-mode to the state machine in McLighting.ino (e.g. after the TV-Mode):

```
  if (mode == WHITELIGHT) {
    whitelight ();
  }
```

3) Add a physical pushbutton and corresponding code, that sets the mode==WHITELIGHT if button is pushed. ( at the very beginning of the main loop() in McLighting.ino):

```
if (button.onPressed()) { //button.onPressed function is from RBDbutton library -Button will be at GPIO12
 mode = WHITELIGHT;
}
```

Would that already be it or did I miss something?

Thanks!
Dimi

PS:
About the A0 usage: 
in this [loop](https://github.com/toblum/McLighting/blob/20ef1b9fac8005997a2034837a7bc400dc0301b1/Arduino/McLighting/McLighting.ino#L376) A0 is used - what is it for?

toblum, 2017-10-25:
From what I can see here, this should work.

Regarding A0: This is just some sample code I forgot. It has no real function. The line https://github.com/toblum/McLighting/blob/20ef1b9fac8005997a2034837a7bc400dc0301b1/Arduino/McLighting/McLighting.ino#L379 and the next can easily be omitted.

closed 2017-10-25 by toblum