toblum/McLighting #52: MQTT Intopic always blank?

MrPasty, 2017-10-29:
Hi,

I am getting this when it tried to connect to my MQTT server?

Attempting MQTT connection 1 / 4 ...
MQTT connected!
MQTT topic in: 
MQTT topic out: 4MLED/out

it repeats this 4 times and ends with:

MQTT connection failed, giving up after 4 tries ...

I am using mosqiotto v3.1.1 - is anyone else experiencing this? I do not have any issues with any of my other MQTT clients.

Thanks
Mr Pasty

banjo1891, 2017-11-02:
I'm having exactly the same problem. Tried using cloudmqtt and mosquitto running on a raspberry pi 2.

toblum, 2017-11-03:
Hi,

I will try this out as soon as I find time, but I did not experience these problems so far. I'm also using mosquitto on a pi.

LeonVos, 2017-11-03:
Do you use user/password for mqtt login/connect, check if this is enebeld !
Gr Leon Vos

MrPasty, 2017-11-03:
I do use a username and password, i added it when i connected to the wifi manager and it appears to have retained it.  I also would have thought if it was username and password issue it would not be able to connect to the out topic but it can.
Thanks

banjo1891, 2017-11-04:
Same here. User/password are enabled, double checked they're both correct and retained. Same result with two different brokers. I was wondering if it could be a firmware issue with the nodemcu. I haven't checked to see if there are any updates, I just got them from eBay and uploaded the mc lighting sketch. The date on them is 2014. I'll have a tinker when I get the chance. Thanks for looking into this.

MrPasty, 2017-11-04:
Pretty sure the problem is not to do with the firmware as I flashed another sketch to mine that also connects to MQTT and that worked fine.

swindi-de, 2017-11-04:
Nur so als Anhaltspunkt....
hier scheint das Problem zu sein. beim drehen der zwei Zeilen passieren auch komische Dinge (aber es tut mit Fehlern).

McLighting.ino -> Zeile 303-304

```ino
String(String(HOSTNAME) + "/in").toCharArray(mqtt_intopic, strlen(HOSTNAME) + 4);
String(String(HOSTNAME) + "/out").toCharArray(mqtt_outtopic, strlen(HOSTNAME) + 5);
```

toblum, 2017-11-05:
Hi @all,

I did some tests with several boards, but could not reproduce the error. I'm not using username&password, but I'm not sure that this is related to this problem.

It seems that just the intopic is not built correctly as @swindi-de mentioned. I'm not sure why only intopic fails.

I committed a few changes, I hope this helps. Could someone that has this error try this out?

banjo1891, 2017-11-05:
Fantastic! Just tested with cloudmqtt and it's now working flawlessly. Thanks again for a great project.

MrPasty, 2017-11-05:
Works great here too!, thanks for fixing this and for such a great project.

closed 2017-11-05 by MrPasty

toblum, 2017-11-06:
Glad that it fixed the problem.

Thank you @swindi-de for pointing at the right direction.