toblum/McLighting #53: MQTT Setup

eps000, 2017-10-29:
Hello, My esp auto connects to my wifi so an ap is never established to input my mqtt info. Is there a way to configure my mqtt host and port info in the ino files?

yanbec, 2017-10-30:
It seems you can set all of the settings in the definitions.h lines 19-34. ;)

toblum, 2017-11-03:
Hi @eps000,

I think @yanbec is right. You can enter it directly there. Or you could start the AP config mode via "http://<mclighting_ip>/start_config_ap". Then connect to the access point of the ESP as described in the wiki.

Regards
Tobias

closed 2017-11-03 by toblum

scordal, 2019-03-06:
Hi guys:

Seems that the device does not connect to MQTT since mine is without username and password. How do I check and correct the MQTT setup and correct it?
I try to access thru the "http://<mclighting_ip>/start_config_ap", but I get a blank screen like following
[image]

Can you please help me to fix it? I am not that literate in thjis, but I am learning a lot.

yanbec, 2019-03-22:
Hey @scordal,
does the problem still persist?
After starting the AP, you should be able to see the access point of the esp and connect to it. Then you can set your configuration like @toblum described above.

Other than that you could still define your MQTT related settings in the definitions.h - in the current version the lines mentioned above have move to 106-109.

If the problem persists, you should maybe open a new issue as this one is quite old.

Good luck!

scordal, 2019-03-23:
yanbed. yes, problem is solved, but at the end I ketp with GPIO. I have also issues related to my Wifi network rejecting more connections (too many wifi devices). Thank you all for the help.
Have a great day.