toblum/McLighting #54: Can I save the last set value?

ChienTing, 2017-11-05:
Can I save the last set value?
ESP-12 automatically reopens after some time
LED light settings will run away after reopening

toblum, 2017-11-06:
Hi @ChienTing,

this is often requested and already on the todo list. I will have a look as soon as I find time. Or maybe someone will share some code.

Regards
Tobias

ptsudza01, 2017-11-10:
Hi @toblum 

I think you can use EEPROM to save value and use timer interrupt to go to function for save something.
I think it can help you 
example:
#include<EEPROM.h>
void timer0_ISR (void){
    // Do some work
    EEPROM.write(0, Count>>8); //Write bit 16 - 9
    EEPROM.write(1, Count&0xFF); //Write bit 8 - 1
    
    timer0_write(ESP.getCycleCount() + 80000000*5); //80Mhz -> 80*10^6 = 1 second
}
void setup() {
    //Read Value from EEPROM 
    Mode |= EEPROM.read(0)<<8; //Read bit 16 - 9
    Mode |= EEPROM.read(1)&0xFF; //Read bit 8 - 1
    
    // put your setup code here, to run once:
    Serial.begin(115200);
    noInterrupts();
    timer0_isr_init();
    timer0_attachInterrupt(timer0_ISR);
    timer0_write(ESP.getCycleCount() + 80000000*5); //80Mhz -> 80*10^6 = 1 second
    interrupts();
}
void loop() {
  // put your main code here, to run repeatedly:

}

Decstasy, 2017-11-10:
I would also love to see this funkctionality. I'm sad that I dont have enough private time to work on it.

toblum, 2017-11-12:
@ptsudza01 Yes, I also thought of this, but did not find time yet to work on it.

toblum, 2018-01-13:
This feature is implemented now. Just update to the latest version.

Regards
Tobias

closed 2018-01-13 by toblum

ptsudza01, 2018-01-17:
@toblum   Thank 👍