toblum/McLighting #56: handleSetSingleLED fails for certain values of LED

matjohn2, 2017-11-22:
While working on a feature to enable ranges of LED's via MQTT, I found some weirdness with LED values over `79` passed to handleSetSingleLED() via MQTT.

At first I thought it was my code, so tried the same via existing `+79ffffff` and `+80ffffff` style notation (also via MQTT) and got the same error. 

Details below.

### A. Fails, my range code, calls `handleSetSingleLED()`.

```
MQTT: Message arrived [R0199ff0000]
MQTT: Set RANGE of LEDS to single color: [R0199ff0000]
Setting RANGE from [1] to [99] as color [ff0000] 
WS: Set single led [1] to [255] [0] [0] (1ff0000)!
WS: Set single led [2] to [255] [0] [0] (2ff0000)!
.. removed for easy reading ..
WS: Set single led [45] to [255] [0] [0] (45ff0000)!
WS: Set single led [46] to [255] [0] [0] (46ff0000)!
WS: Set single led [47] to [255] [0] [0] (47ff0000)!
.. removed for easy reading ..
WS: Set single led [78] to [255] [0] [0] (78ff0000)!
WS: Set single led [79] to [255] [0] [0] (79ff0000)!
WS: Set single led [80] to [255] [255] [255] (80ff0000)!
WS: Set single led [81] to [255] [255] [255] (81ff0000)!
WS: Set single led [82] to [255] [255] [255] (82ff0000)!
WS: Set single led [83] to [255] [255] [255] (83ff0000)!
WS: Set single led [84] to [255] [255] [255] (84ff0000)!
WS: Set single led [85] to [255] [255] [255] (85ff0000)!
WS: Set single led [86] to [255] [255] [255] (86ff0000)!
WS: Set single led [87] to [255] [255] [255] (87ff0000)!
WS: Set single led [88] to [255] [255] [255] (88ff0000)!
WS: Set single led [89] to [255] [255] [255] (89ff0000)!
WS: Set single led [90] to [255] [255] [255] (90ff0000)!
WS: Set single led [91] to [255] [255] [255] (91ff0000)!
WS: Set single led [92] to [255] [255] [255] (92ff0000)!
WS: Set single led [93] to [255] [255] [255] (93ff0000)!
WS: Set single led [94] to [255] [255] [255] (94ff0000)!
WS: Set single led [95] to [255] [255] [255] (95ff0000)!
WS: Set single led [96] to [255] [255] [255] (96ff0000)!
WS: Set single led [97] to [255] [255] [255] (97ff0000)!
WS: Set single led [98] to [255] [255] [255] (98ff0000)!

```
As you can see, LED's after 80 ignore the RGB value provided in `payload` to the `handleSetSingleLED()` function.

### B. Fails, Direct MQTT calls to `handleSetSingleLED()`.
```
MQTT: Message arrived [!79ff0000]
WS: Set single led [79] to [255] [0] [0] (!79ff0000)!
MQTT: Set single LED in given color [!79ff0000]
MQTT: Message arrived [!80ff0000]
WS: Set single led [80] to [255] [255] [255] (!80ff0000)!
MQTT: Set single LED in given color [!80ff0000]
```

Also, I think potentially there's a memory corruption/leak going on somewhere (maybe a different problem); when exercising the `handleSetSingleLED()` function repeatedly for a number of LED's and values, some LED's get 'stuck' on a random color.

Setting those LED's to white `FFFFFF` seems to work, but other commands fail (ie, still sometimes 'stuck', just now white.)

toblum, 2017-11-22:
Hi @matjohn2,

thank you for your feedback. 
I thought I had this fixed with this commit: https://github.com/toblum/McLighting/commit/e09957d0324f33691a995efe343f6f951777f1d7 
Do you use the latest version?

I will also have a look at your PR as soon as I find time.

Regards
Tobias

matjohn2, 2017-11-22:
Hi @toblum 
Awesome project by the way! I'm running this code from the current master here: https://github.com/toblum/McLighting/blob/master/Arduino/McLighting/request_handlers.h#L70

it's slightly different to commit e09957d in that we're passing in `firstChar` to the function instead of having that value set explicitly to `1`?

matjohn2, 2017-11-23:
So this could turn into a bit of a rant, but I spent some time looking at this last night.

I'm not (at all!) a C programmer or hardware/embedded systems guy, so theres potential for a lot of error in my findings.

Problem: LED's >= 80 become the wrong color and get 'stuck':

```
MQTT: Message arrived [R7599ff0000]
Setting RANGE from [75] to [99] as color [ff0000]
WS: Set single led [77] to [255] [0] [0] (77ff0000)!
WS: Set single led [78] to [255] [0] [0] (78ff0000)!
WS: Set single led [79] to [255] [0] [0] (79ff0000)!
WS: Set single led [80] to [255] [255] [255] (80ff0000)!
WS: Set single led [81] to [255] [255] [255] (81ff0000)!
WS: Set single led [82] to [255] [255] [255] (82ff0000)!
WS: Set single led [83] to [255] [255] [255] (83ff0000)!
WS: Set single led [84] to [255] [255] [255] (84ff0000)!
WS: Set single led [85] to [255] [255] [255] (85ff0000)!
```

Turning on some debugging, I printed the rgb long int variable for each LED:

```
WS: Set single led [77] to [255] [0] [0] (77ff0000)!
DEBUG: rgb: [2029977600] tmpled: [<null>] led: [1073687488]
WS: Set single led [78] to [255] [0] [0] (78ff0000)!
DEBUG: rgb: [2046754816] tmpled: [<null>] led: [1073687488]
WS: Set single led [79] to [255] [0] [0] (79ff0000)!
DEBUG: rgb: [2147483647] tmpled: [<null>] led: [1073687488]
WS: Set single led [80] to [255] [255] [255] (80ff0000)!
DEBUG: rgb: [2147483647] tmpled: [<null>] led: [1073687488]
WS: Set single led [81] to [255] [255] [255] (81ff0000)!
DEBUG: rgb: [2147483647] tmpled: [<null>] led: [1073687488]
WS: Set single led [82] to [255] [255] [255] (82ff0000)!
DEBUG: rgb: [2147483647] tmpled: [<null>] led: [1073687488]
WS: Set single led [83] to [255] [255] [255] (83ff0000)!
DEBUG: rgb: [2147483647] tmpled: [<null>] led: [1073687488]
WS: Set single led [84] to [255] [255] [255] (84ff0000)!
DEBUG: rgb: [2147483647] tmpled: [<null>] led: [1073687488] 
```

(This is where i could be very mistaken) but I was expecting to see a char of the LED number in `tmpled`, a two digit int in `led` and the decimal notation of the color values in `rgb`.

The `rgb` values struck me as odd, as they should just contain color, which should be the same for all these LED's (based on the range i'd set). The fact that each LED seems to have an incrementing `rgb` value was weird.

Also, I noticed that LED number 80 is when the value gets to `2147483647`, this is the maximum value for an INT of it's size, so each LED after this stays at the same RGB value, there's no room for further increase. 

I then double checked what the next step was expecting;

```
  strip.setPixelColor(led, ledstates[led].red, ledstates[led].green, ledstates[led].blue);
  strip.show();
```

or

```
  strip.setPixelColor(i, ledstates[i].red, ledstates[i].green, ledstates[i].blue);
  strip.show();
```

I confirmed by manually setting these to different values that the code expected regular int's for the LED's 0-99 and 0-255 values for R, G and B.

I couldnt fully grok (it was late, and I haven't touched bitwise shifts since compsci) why exactly I was seeing such large, growing numbers for the LED and RGB values, so i decided to re-write `handleSetSingleLED()` to prove to myself that the large number were the issue.

```
void handleSetSingleLED(uint8_t * mypayload, uint8_t firstChar = 0) {
  // decode led index
  char templed[3];
  strncpy (templed, (const char *) &mypayload[firstChar], 2 );
[... 63 lines cut ...]
WS: Set single led [85] to [0] [0] [255] (850000ff)!
rgb.red: [0] rgb.green: [0] rgb.blue: [255]
WS: Set single led [86] to [0] [0] [255] (860000ff)!
rgb.red: [0] rgb.green: [0] rgb.blue: [255]
WS: Set single led [87] to [0] [0] [255] (870000ff)!
rgb.red: [0] rgb.green: [0] rgb.blue: [255]
WS: Set single led [88] to [0] [0] [255] (880000ff)!
rgb.red: [0] rgb.green: [0] rgb.blue: [255]
WS: Set single led [89] to [0] [0] [255] (890000ff)!
rgb.red: [0] rgb.green: [0] rgb.blue: [255]
WS: Set single led [90] to [0] [0] [255] (900000ff)!
rgb.red: [0] rgb.green: [0] rgb.blue: [255]
WS: Set single led [91] to [0] [0] [255] (910000ff)!
rgb.red: [0] rgb.green: [0] rgb.blue: [255]
WS: Set single led [92] to [0] [0] [255] (920000ff)!
```

I tested with the following command (to MQTT) overnight (using the range PR in #55)

```
while [ 1 == 1 ] ; do for i in R0030ff0000 R316000ff00 R61990000ff R003000ff91 R31600000ff R6199ff3c2d ; do mosquitto_pub -h 10.22.0.197 -t "ESP8266_01/in" -m $i ; sleep 0.5 ; done ; done
```

Video of the commands running here: [](https://www.youtube.com/watch?v=fVlmnXkaWlk
)

toblum, 2017-11-25:
Hi @matjohn2,

thank you for your investigations and and contributions. I did not test strips with more than 60 LEDs because I don't own such big strips. I'm also no C developer (coming from the webdev side) so I'm not the one to judge on good or bad C code. :-)
Your implementation looked good to me and at least works with my 60 LED strip, so I took it and merged it into the master branch.

Best regards
Tobias

closed 2017-11-25 by toblum