toblum/McLighting #59: MQTT not working

JyriT, 2017-11-27:
Any tips what to do next? 

MQTT server working without authentication, another MQTT client sends data ok. 

Log From serial debug:
Attempting MQTT connection 1 / 4 ...
MQTT connected!
MQTT topic in: ESP/in
MQTT topic out: ESP/out
Attempting MQTT connection 2 / 4 ...
MQTT connected!
MQTT topic in: ESP/in
MQTT topic out: ESP/out
Attempting MQTT connection 3 / 4 ...
MQTT connected!
MQTT topic in: ESP/in
MQTT topic out: ESP/out
Attempting MQTT connection 4 / 4 ...
MQTT connected!
MQTT topic in: ESP/in
MQTT topic out: ESP/out
MQTT connection failed, giving up after 4 tries ...

and from mosquite stream:
Client mosqsub/13739-terramstr received PUBLISH (d0, q0, r0, m0, 'ESP/out', ... (21 bytes))
McLighting ready: ESP
Client mosqsub/13739-terramstr received PUBLISH (d0, q0, r0, m0, 'ESP/out', ... (21 bytes))
McLighting ready: ESP
Client mosqsub/13739-terramstr received PUBLISH (d0, q0, r0, m0, 'ESP/out', ... (21 bytes))
McLighting ready: ESP
Client mosqsub/13739-terramstr received PUBLISH (d0, q0, r0, m0, 'ESP/out', ... (21 bytes))
McLighting ready: ESP
Client mosqsub/13739-terramstr received PUBLISH (d0, q0, r0, m0, 'esp/in', ... (3 bytes))
=tv
Client mosqsub/13739-terramstr received PUBLISH (d0, q0, r0, m0, 'esp/in', ... (3 bytes))
=tv
Client mosqsub/13739-terramstr received PUBLISH (d0, q0, r0, m0, 'esp/in', ... (3 bytes))
=tv

giddyhup, 2017-12-02:
Do you perhaps have other MQTT clients with the same client ID (ESP)? Brokers don't like that. Change the `const char mqtt_clientid[]` in definitions.h and try it with that.

toblum, 2017-12-02:
Hi @JyriT 
From the log it looks fine to me except for the reconnects. I didn't experience this behaviour so far. @giddyhup idea is good, that might be a problem.

JyriT, 2017-12-03:
Fixed! Hostname change did it. Case can be closed now. 
Thanks!

toblum, 2017-12-03:
Great!

closed 2017-12-03 by toblum