toblum/McLighting #60: Issue with ranges and colors

giddyhup, 2017-12-02:
I've entered the proper number of LEDs in the code (`#define NUMLEDS 240`), I've tried it via MQTT and via websockets. I've set the proper color order (the color wheel and effects confirm that). Effects work as expected but addressing LED ranges fails.  

First of all, the syntax seems to assume that there are not more than 100 LEDs. Yet, even then the payload `R0009ffffffR1019ff0000` turns the first 10 LEDs white and the next 10 yellow (not red). Am I missing something?

toblum, 2017-12-02:
Hi @giddyhup,

the limit of 100 LEDs was made by me because I don't own such long strips and didn't think of that when I first implemented it. Then I don't wanted to change it because I didn't want to break someones implementation. But it should be easy to allow for 1000 LEDs by patching the code in handleRangeDifferentColors().

It's strange with the wrong color. What output do you get on serial when issuing the command? That code was implemented by @matjohn2 not long ago, maybe he has an idea

giddyhup, 2017-12-03:
@toblum Thanks for the quick response. I'll look how far I can get with the function and regarding the color I'll investigate further when I'm done with the amount of LEDs.

giddyhup, 2017-12-03:
I was able to modify the code to work with ranges up to 1000 LEDs. Regrettably, I ran into new issues with MQTT/pubsub. There is a limit for payload length. This payload is too long: `R000019ff0000R02003900ff00R040059ff0000R06007900ff00R080099ff0000R10011900ff00R120139ff0000R14015900ff00R160179ff0000R18019900ff00R200219ff0000R22023900ff00`.

giddyhup, 2017-12-03:
Solved by modifying PubSubClient.h:

```
// MQTT_MAX_PACKET_SIZE : Maximum packet size
#ifndef MQTT_MAX_PACKET_SIZE
// #define MQTT_MAX_PACKET_SIZE 128
#define MQTT_MAX_PACKET_SIZE 254
#endif

```

closed 2017-12-03 by giddyhup

giddyhup, 2017-12-03:
These are my modifications in request_handlers.h
```

void handleSetSingleLED(uint8_t * mypayload, uint8_t firstChar = 0) {
// decode led index
// !! adapted for three digit index 
// mod
  char templed[4];
  strncpy (templed, (const char *) &mypayload[firstChar], 3 );
// end mod
(...)
// mod    
    strncpy (redhex, (const char *) &mypayload[3], 2 );
    strncpy (greenhex, (const char *) &mypayload[5], 2 );
    strncpy (bluehex, (const char *) &mypayload[7], 2 ); 
// end mod
(...)
void handleRangeDifferentColors(uint8_t * mypayload) {
  uint8_t* nextCommand = 0;
  nextCommand = (uint8_t*) strtok((char*) mypayload, "R");
  //While there is a range to process R001010<00ff00>

  while (nextCommand) {
      //Loop for each LED.
// mod
      char startled[4] = { 0,0,0,0 };
      char endled[4] = { 0,0,0,0 };
      char colorval[7] = { 0,0,0,0,0,0,0 };
      strncpy ( startled, (const char *) &nextCommand[0], 3 );
      strncpy ( endled, (const char *) &nextCommand[3], 3 );
      strncpy ( colorval, (const char *) &nextCommand[6], 6 );
// end mod
(...)
// mod
          char rangeData[11] = { 0,0,0,0,0,0,0,0,0,0,0 };
          if ( rangebegin < 10 ) {
            //Create the valid 'nextCommand' structure
            sprintf(rangeData, "00%d%s", rangebegin, colorval);
          }
          if ( rangebegin >= 10 ) {
            //Create the valid 'nextCommand' structure
            sprintf(rangeData, "0%d%s", rangebegin, colorval);
          }
          if ( rangebegin >= 100 ) {
            //Create the valid 'nextCommand' structure
            sprintf(rangeData, "%d%s", rangebegin, colorval);
          }
// end mod
```

matjohn2, 2017-12-03:
Hey @giddyhup 
Only just seen this.

Ironically, I added the 'range' command because I have 99 LED's and I didn't want to have to have a really long  MQTT string for the individual LED control notation, (+01ffffff+02ffffff etc).

Did the colour problem go away with the above changes? 
Thanks for updating handleRangeDifferentColors() with a three char ID as well.

giddyhup, 2017-12-03:
Hi @matjohn2 

Yep. The color issue went away as well. rangeData needed to be adjusted from 9 to 11. Yet now I am facing issues with stability. My ESP8266 becomes unresponsive and restarts.

I think JSON would be better suited for pattern setting. But pubsub doesn't seem to be suited for more payloads.

Tschabenga, 2022-01-04:
> I was able to modify the code to work with ranges up to 1000 LEDs. Regrettably, I ran into new issues with MQTT/pubsub. There is a limit for payload length. This payload is too long: [...]

You can manage up to 256 (0-255) LEDs with your modifications which is great. With a small modification in `handleSetSingleLED` you really can manage ub to 1000 LEDs by simply replacing
`uint8_t led = atoi(templed);`
with 
`uint16_t led = atoi(templed);`