toblum/McLighting #64: WS2812 Going funny

Lamoresk, 2017-12-05:


AmpacheUser, 2017-12-05:
I experience the same exact issues as well.

AmpacheUser, 2017-12-06:
OK, After looking into this a little bit more, there _are_ a few effects that do allow me to control the speed (but seems to work opposite as you would expect, 255 is the slowest, 0 is the fastest), but most of them do not allow me to change the speed. The ones that work so far are, Blink, Random Color, Single Dynamic, Multi Dynamic, Sparkle, Flash Sparkle, Hyper Sparkle, Strobe, Strobe Rainbow, Blink Rainbow, Fire Flicker, and Fire Flicker (soft). All others you cannot control the speed.

toblum, 2017-12-08:
Hi @Lamoresk and @AmpacheUser,

what do you get at the serial console when you change the speed? I didn't experience such behaviour so far.

Regards
Tobias

AmpacheUser, 2017-12-09:
Does not show any errors and the output is the same as when it is working. 
Here is some output from the serial console:

WS: [0] get Text: /18
WS: [0] get Text: ?2
WS: Set speed to: [2]
WS: [0] get Text: ?253
WS: Set speed to: [253]
WS: [0] get Text: ?10
WS: Set speed to: [10]
WS: [0] get Text: ?52
WS: Set speed to: [52]

RBOUMAN, 2017-12-10:
The Speedparameter in w2812Fx has changed, it now has a value from 0(fastest)-65535 (slowest). The slider speed in Mc Lighing's index,htm delivers a value from 0 to 255, so from atomic speed downto a very fast speed. The colors of Christmas red and green mix up so do you see yellow :). It needs a value of approx 10000 to animate smooth. I modified the code for the slider and some code for the speedvariable (uint16 instead of uint8) in the code and i my version the speed can be controlled in a decent way
WS: [0] get Text: /47
WS: [0] get Text: ?10918
WS: Set speed to: [10918]

AmpacheUser, 2017-12-11:
@RBOUMAN That explains it then. Looks like McLighting will have to be modified to support this.
For now I used the development branch of WS2812FX as it is behind the master. You lose a bunch of effects but you can control the speed of them all.

toblum, 2017-12-12:
Hi,

thanks for your investigations. I will have to try with the latest WS2812FX. I think I should convert the McLighting speed (0-255) to the new WS2812FX speeds (65535-0) to keep API for McLIghting consistent.
I will look into this as soon as I find time. Or maybe someone wants to contribute code here.

toblum, 2017-12-16:
I released a new version that works with the new WS2812FX speeds:
https://github.com/toblum/McLighting/commit/2e58d55eec917d08b672e4f0b98d49669a2d5634

I hope that every thing works now again as expected.

closed 2017-12-16 by toblum