toblum/McLighting #66: How to change green and blue ?

DimIsaev, 2017-12-10:
Strip  WS2811 IC   Leds 

Confused green and blue light
How to change green and blue ?

toblum, 2017-12-12:
Hi,

is this a standard WS2811 strip or a special model? I never had such issues,

Regards
Tobias

DimIsaev, 2017-12-13:
hi  
WS8211ic

https://ru.aliexpress.com/item/best-price-5m-DC12V-ws2811ic-5050-RGB-SMD-individually-addressable-ws2811-led-pixels-strip/32428217949.html

toblum, 2017-12-18:
This looks like a standard WS2811.

What happens when you try one of these standard examples?
https://github.com/adafruit/Adafruit_NeoPixel/tree/master/examples

Are the colors then OK?

LeonVos, 2017-12-19:
Maybe change "GRB" "BRG" "GRB" "RGB" or "RBG" in code ??

dcuth79, 2017-12-23:
I have the same problem as DimIsaev.  The IC is WORLDSENI ws2811.  Where in the code can you change like LeonVos said?  Only line I can find that change the led is.

WS2812FX strip = WS2812FX(NUMLEDS, PIN, NEO_GRB + NEO_KHZ400).

But only takes NEO_RGB and that makes red=blue green=red and blue=red.

toblum, 2017-12-24:
@dcuth79 Yes, the is the mentioned line, It takes the values as listed in the Adafruit Neopixel library: https://github.com/adafruit/Adafruit_NeoPixel/blob/fe64029c50d94492060aab58896cd570d0ac6226/Adafruit_NeoPixel.h#L54

So NEO_BRG (and all other permutations) should also be possible.
Have also a look here: https://learn.adafruit.com/adafruit-neopixel-uberguide/arduino-library-use

closed 2018-01-13 by toblum