toblum/McLighting #67: Static ip wifi and mqtt

lampenschakelen, 2017-12-17:
Thanks again for making this wonderful program. 

I like to make alarm wake up light. 
With node-red and connect it with MQTT. 

I like to change it the ESP8266 program to static IP and the MQTT IP address and password but I don't understand where I have to fill this in.   There is no configuration page. 

The code is specially designed with Wifimanager.
It is possible to setup a static IP but how?
https://github.com/tzapu/WiFiManager/tree/master/examples/AutoConnectWithStaticIP

I hope somebody can help me out. 
If someone has already some setup in node-red please share you're code.

Thanks a lot already,

toblum, 2017-12-18:
Hi @lampenschakelen,

you're welcome. I never tried static IPs, but I think you should insert the lines with:
IPAddress _ip = IPAddress(10, 0, 1, 78);
IPAddress _gw = IPAddress(10, 0, 1, 1);
IPAddress _sn = IPAddress(255, 255, 255, 0);
wifiManager.setSTAStaticIPConfig(_ip, _gw, _sn);

Somewhere around line:
https://github.com/toblum/McLighting/blob/b04896700eb065e0e9c23ddbb70b2202aa916dfb/Arduino/McLighting/McLighting.ino#L218

Maybe you could give this a try.

Regards
Tobias

lampenschakelen, 2017-12-18:
Where i have to fill in the mqtt ip and password? I like to do the connection with node-red.  And setup a timmer to make the wake me up

lampenschakelen, 2017-12-18:
Thanks a lot @toblum static ip is working now ;-)  It's not possible to go back to wifimanager and change the settings? From the user interface.  The last setup is the mqtt i hope you can help me with that.  I need a fill in server ip  / username / password /  poortnumber ecta ... Maybe you can share some nice Node-red codes to test .

lampenschakelen, 2017-12-18:
Thanks a lot @toblum for helping. 

I found the following page: 
https://github.com/toblum/McLighting/wiki/HTTP-REST-API 

This working perfectly: 
http://Ip-adres-of-your-esp/reset_wlan 
* Disconnect you're lights and reboot your esp first an hit enter. 
Now you can setup easily MQTT server. 

**@toblum maybe good idea to make config page? To change wifi / mMQTT server / ip-adress options **

toblum, 2017-12-20:
Hi @lampenschakelen,
great you found a solution. 
"/start_config_ap" should also work.

Regards

closed 2017-12-20 by toblum