toblum/McLighting #68: Wemos D1 Mini - Watchdog Reset

canadiansteve, 2017-12-20:
I have had McLighting working great with a couple of ESP8266-01 modules.  I have 3 new Wemos D1 Mini's and figured it would be a pretty easy switch over.

Unfortunately, I can install the software, configure an access point, get to the point of having the webserver go live and then Watchdog crashes the device and, even with a powerdown, it will not reboot until reflashed.

The serial traffic indicates:

----------

WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.130
connected...yeey :)
Open http://192.168.1.130/ to open McLighting.
Use http://ESP8266_01.local/ when you have Bonjour installed.
New users: Open http://192.168.1.130/upload to upload the webpages first.

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

----------

From the research I have been doing, this appears to be a H/W issue but I cannot seem to identify any source of issue. Does anyone have experience getting McLighting to work on the Wemos chips?

Thanks.

giddyhup, 2017-12-20:
I flashed a Wemos d1 mini from own-written code to McLighting OTA so I didn't see the serial output but I also experienced resets. Now I use [this](https://github.com/bruhautomation/ESP-MQTT-JSON-Digital-LEDs) and it works.

canadiansteve, 2017-12-22:
I seem to have resolved the issue and suspect there was some flash corruption in a previous upload that McLighting was not overwriting.  After, first, identifying that I likely have a Chinese knockoff WeMos and that my flash is only 1MB rather than the expected 4MB, I was able to clear the flash entirely and reload McLighting without issue.

Thank you to all the devs for the excellent software.

toblum, 2017-12-22:
Great that you made it work. I also had problems with some bad quality wemos boards in another project. Newer wemos boards also often need the DIO flash mode. 
Regards 
Tobias

closed 2017-12-22 by toblum

lolomad, 2018-11-25:
Hello, 
I have the same issue with my WEMOS. I program with arduino IDE and put "wemos D1 R1".
The output (serial.monitor ) are not ok. And the selection from the web page  works but the leds does not changed.
So I swith to use nodemcu from the arduino IDE and at the moment the trace in serial.monitor are ok.
But I can not see my leds changing also.

So  what I have to do ? to clean the firmware? The wemos is new .  I tried with 2 wemos. Same result. They are chinese from aliexpress. so clean the firmware and write only in    1M. But how to do this.
I use wemos D1 R1. Do I have to change. I can see also LOLIN (Wemos D1...).
I need some tips.
Thanks again for this fantastic program. I did some months ago another project with your program. Now it is  to install in the kitchen.

Laurent