toblum/McLighting #69: File Not Found

stoehrmark, 2017-12-22:
After uploading to ESP8266 then reboot and connecting  to WIFI with WiFiManager no problem I get the following
~~~~~~~~~~~~~~~~~~~~~~~~~~~~
File Not Found

URI: /
Method: GET
Arguments: 0
~~~~~~~~~~~~~~~~~~~~~~~~~~~
Have you seen this before?

stoehrmark, 2017-12-22:
SORRY, I did not follow the WiKi and upload the index.htm file. All good now. EXCELLENT work thank you. My kids are going to love the lamps

Star from Ikea (STRALA)
ESP8266,  WS2812 and very small 5-3.3V power supply inside bubble gum container 
(PS from [http://www.electrodragon.com/product/ac-85-265v-dc-5v-power-module/](url)) Works Great but must add AMS1117 for 3.3V
[image]
[image]
[image]
[image]
[image]
[image]

toblum, 2017-12-22:
Great that you figured it out. 
BTW: very nice build. 

Regards
Tobias

closed 2017-12-22 by toblum