toblum/McLighting #70: No ESP8266_01 network

zangaby, 2017-12-22:
Hi,

Just flashed the sketch on a nodemcu and on a wemos d1 mini, unfortunately not ESP8266_01 network is visible on first boot.
 What could I be missing?

Thanks !

lampenschakelen, 2017-12-23:
Hi Zangaby, 

Follow the steps: 
https://youtu.be/T1Ul7iN72lI?t=243
Our check the full video. 

It has to work!

Best regards, 
Lampenschakelen

zangaby, 2017-12-24:
Hi Lampenschakelen,

Thank you for your reply!
I used these boards with a different project before and it seems that the credentials were not overwritten because I noticed the board was automatically connecting to the AP (thanks for the video where it was showing that it's going thru the rainbow effect when connected).

Marry Christmas !

Best regards,
zangaby

closed 2017-12-24 by zangaby