toblum/McLighting #71: HTTP API set_brightness not working

athphane, 2017-12-25:
Using WS2812b LED strip with NODEMCU Amica.

Yeah set_brightness not working. I get zero brightness. No matter the value I use.

http://192.168.1.11/set_brightness?p=255

Doesn't work.

Help. Or should I use Websockets or MQTT?

LeonVos, 2017-12-25:
Try    p= with range from 0 to 100

athphane, 2017-12-25:
Doesn't work. Still zero brightness..
Here's the output.
`{"mode":3, "ws2812fx_mode":12, "ws2812fx_mode_name":"Rainbow Cycle", "speed":255, "brightness":50, "color":[0, 0, 0]}`

LeonVos, 2017-12-25:
You have not set a color   = 0.0.0 , so there is no light any way even with b=100 there will be NO light.
Put after your brightness code  ?r=255,g=255,b=255  and you will get white light.

athphane, 2017-12-25:
How do I do that? Example URL please?

LeonVos, 2017-12-25:
for a color (RED)  
http://192.168.192.160/all?r=255&g=0&b=0
And for brightness  (use C !  :-)
http://192.168.192.160/set_brightness?c=10

athphane, 2017-12-26:
So set_brightness only works for manually set colors? I'm trying to control brightness of the included light effects.

LeonVos, 2017-12-26:
Should work

LeonVos, 2017-12-26:
Look here for all the api options.
https://github.com/toblum/McLighting/issues/71#issuecomment-353929935

LeonVos, 2017-12-26:
This one :-)

https://github.com/toblum/McLighting/wiki/HTTP-REST-API

athphane, 2017-12-26:
Alright so this is basically how my code runs.

192.168.1.11/set_mode?m=11
192.168.1.11/set_speed?d=255
192.168.1.11/set_brightness?c=10

And I get no brightness.. 
Why tho?

LeonVos, 2017-12-26:
Try c=100  (is max)

And what is the outcome?

athphane, 2017-12-26:
Darkness

LeonVos, 2017-12-26:
But did you set a color ??
Mode 11 needs a color to be set ! 
Test it with mode 8, this will work with your 'code'

athphane, 2017-12-29:
3 days later...
Still cannot get the brightness to set. Kinda...
When I go to /set_brightness?c=100 it goes to a 100% brightness and goes to 0% brightness. In less than a second. 
Pls help.

LeonVos, 2017-12-29:
@athphane, :-)
I did now test it with your code order.
It seems like set_brightness also sets to static ??
So , try this orde
http://192.168.1.11/set_brightness?c=100  
192.168.1.11/set_mode?m=9
192.168.1.11/set_speed?d=255

Should work for you, for some modes you need to set a color (like scan mode_11needs a color)
Good luck

athphane, 2017-12-29:
@LeonVos 
Thank you very much. That order worked perfectly fine. 

> for some modes you need to set a color (like scan mode_11needs a color)

Umm, how do I set the color? There is no set_color in the documentation...
Also, do I set the color after I set the brightness?

LeonVos, 2017-12-29:
http://192.168.1.11/all?r=255&g=255&b=255
(=white)
1ste all the value's, then the mode.

Your welcome

closed 2018-01-13 by toblum