toblum/McLighting #74: File not found even after uploading the index.htm file

khumps, 2017-12-27:
I have gotten to the step in the software instructions that requires me to upload the index.htm file to the board. I upload it as such https://i.imgur.com/ndIQsWf.png, it then brings me to a blank /edit page https://i.imgur.com/YPpDUm2.png, i go back to the root site and i get a file not found https://i.imgur.com/q5Z4sFz.png. I have read the directions over multiple times but i could just be missing something.
Thanks,
Kevin

toblum, 2017-12-27:
This looks correct to me. Don't know why it shows "file not found".

It you upload https://github.com/toblum/McLighting/blob/master/Arduino/McLighting/data/edit.htm.gz and call http://mclighting_ip/edit
What does it show?

khumps, 2017-12-27:
Nope, just a file not found page :/ https://i.imgur.com/kkHHqLr.png

toblum, 2017-12-28:
What is your setting for "Flash size" in the Arduino IDE? What SPIFFS size did you choose?

khumps, 2017-12-29:
I actually had a ton of issues getting the Arduino IDE to flash properly. I ended up using esptool with all settings as default. I am going to take another stab at the arduino IDE and see if I can get it to work

toblum, 2017-12-29:
That could be an issue. Some part of the flash should be reserved for the SPIFFS filesystem.

khumps, 2017-12-30:
I get it to flash with these options https://i.imgur.com/BrwzevL.png. It finishes flashing, I get it to the WifiManager setup, i put in wifi credentials, it "reboots" but it is not visible on the network.

toblum, 2017-12-30:
This looks completely OK. What does the serial monitor show when that happens? 
I usially use these settings: https://imgur.com/a/xXlf2 but yours should be ok.

khumps, 2017-12-30:
That's another issue in itself. I have not once gotten any output from the serial monitor.

toblum, 2017-12-30:
This looks like problems with flashing. Did you try to set Flash mode to DOUT? Which board type do you use?

khumps, 2017-12-30:
Board is ESP8266-01

toblum, 2017-12-30:
Ah, that'a a important information. If I remember correctly (haven't used a pure ESP8266-01 for a long time) it only has 1 MB of flash, so you should be using a setting like "1M (256 K SPIFFS)".

khumps, 2017-12-30:
Thank you, that was the issue. That solved everything. Now the last 7 leds of the 30 led strip still arent working but i have no way to prove that they were not DOA. I have changed the NUMLEDS property to 30 under definitions.h

toblum, 2017-12-31:
If you configured the NUMLEDs correctly, it should work for all LEDs. It could be that one LEDs is broken and that kills the rest of the line. I had that too.

The ESP8266 behaves a bit strange in your case. I did not experience this i the past, but I always used NodeMCU boards with 4 MB of memory. How do you power it?

toblum, 2017-12-31:
Have a look here regarding serial debug:
https://github.com/toblum/McLighting/issues/21#issuecomment-354609072

khumps, 2018-01-02:
Everything is fixed now, That fixed the serial issue. Also, my strip is an RGBW strip so changing the type fixed all color issues as well as getting all the LED's functioning. I am also using the latest version of the Neopixel library (1.1.3) Thank you for your help

toblum, 2018-01-02:
Great! Have fun with it!

closed 2018-01-02 by toblum