toblum/McLighting #75: NodeMCU random restart..

athphane, 2017-12-29:
Has this got anything to do with the McLighting software?
It restarts in the middle of the night and when I wake up it's turned on with the default . I do not want to waste electricity.
10:30PM it restarted.
10:52PM IT RESTARTED AGAIN!!!

athphane, 2017-12-29:
Never mind.. Should've read the other issues. Feel so stupid..

closed 2017-12-29 by athphane

davidblumer, 2018-07-02:
What exactly was the problem here? i'm facing the same issues

toblum, 2018-07-03:
Hi @davidblumer,

In this case I don’t know what’s the issue. But we noticed instabilities with the WS2812FX library in some setups. @debsahus animation library seems not to have these issues. What library are you using?
Also always double check your power supply. 
Regards 
Tobias

athphane, 2018-07-04:
I started using my lights in May after 4 months of inactivity. Flashed the newest version of McLighting at the time and it is working without a hitch. 
Same NodeMCU board I used months back, No problems since.