toblum/McLighting #76: ESP_HOST unreachable after x amount of hours.

athphane, 2018-01-01:
Going to the IP address of the NodeMCU module does not load the web interface. 
Chrome tells me "192.168.1.5 took too long to respond."
This usually happens when I send a bunch of commands through the HTTP API, and then it becomes unresponsive.
Keep in mind this still happens even when I am not using the HTTP API. Just a few hours later it becomes unreachable. The LED strip still works.
Tried this with another NodeMCU that I have and it still does not work.
Help.

toblum, 2018-01-02:
This behaviour is hard to track. Could you see whats on the serial monitor when that happens?

athphane, 2018-01-02:
Gimme a few to get the problem happening again. I will try to force it by giving a bunch of commands.
Updates coming soon.. Please bear with me..

athphane, 2018-01-03:
I don't think that this is gonna help at all, but here's the serial output when I cannot reach the webgui.

`⸮⸮=r⸮⸮⸮s⸮⸮⸮⸮!⸮M⸮⸮h>/⸮-⸮⸮⸮⸮`

Bunch of gibberish. Nothing human readable.

toblum, 2018-01-13:
Hi @athphane,

thanks for the feedback. I had similar problems with a system before where the power supply wasn't stable enough. The ESP is quite sensible with that. How do you power it?
Some people reported also boards with bad flash memory to behave that way.

Regards
Tobias

athphane, 2018-01-14:
I have connected my 5V 10A power supply by soldering the positive and negative leads to the bare leads of the LED strip, and then I have jumper wires going from the "jack" of the LED strip. 

[image]

Like that but mine has 3, not 4. 
5v, Data and GND.

But now since you mentioned it, it may be a problem of the flash memory of the the board. I did buy 2 of them for $3 each on aliexpress. And yes I flashed the other one with it and the same problem persists. Guess I should look into getting reliable ones. 

Got any recommendations on other sites I can get them from, or maybe a eBay seller that you have purchased from before?

Thanks
Athfan

lukasKaiser, 2018-01-17:
@athphane I had the same issue but once I switched the baud rate from 9600 to the propper 115200 I had no issues any more 😉 

regards Lukas

athphane, 2018-01-18:
I've again placed another order for the ESP modules. This time the WEMOS D1 mini. Trying my luck. And @lukasKaiser, i will try to flash it like you said, but right now I'm away from home on holiday. Will do as soon as I get back home.

kotelmach, 2018-02-10:
I’ve been using Wemos D1 boards. If I supply the board with 5v, I notice the WiFi is very unstable. The ESP boards seem to need 12v while the lights only need 5v.

toblum, 2018-02-10:
Hi @kotelmach,

it depends on the board. If you're using Standard Wemos D1 boards and power them via jack, the input voltage is 9-24 V, so 5V are surely not enough, especially when the WiFi is active. You can power these boards with 12V and user their 5V 1A output depending on the number of LEDs.

Regards

closed 2018-02-10 by toblum