toblum/McLighting #77: Error compiling for board nodemcu

e11i0t23, 2018-01-06:
The error message is below sorry about formatting it is a copy and paste from arduino ide any suggestions on how to fix are welcome

Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Prebuilt (MSS=536), Disabled, None, 115200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Elliot\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Elliot\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Elliot\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\Elliot\AppData\Local\Temp\arduino_build_669581 -warnings=none -build-cache C:\Users\Elliot\AppData\Local\Temp\arduino_cache_918768 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.12 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Users\Elliot\Downloads\McLighting-master\Arduino\McLighting\McLighting.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Elliot\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Elliot\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Elliot\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=80,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,UploadSpeed=115200 -ide-version=10805 -build-path C:\Users\Elliot\AppData\Local\Temp\arduino_build_669581 -warnings=none -build-cache C:\Users\Elliot\AppData\Local\Temp\arduino_cache_918768 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.12 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Users\Elliot\Downloads\McLighting-master\Arduino\McLighting\McLighting.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0
Using core 'esp8266' from platform in folder: C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0
Detecting libraries used...
"C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Elliot\AppData\Local\Temp\arduino_build_669581/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\nodemcu" "C:\Users\Elliot\AppData\Local\Temp\arduino_build_669581\sketch\McLighting.ino.cpp" -o "nul"
"C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Elliot\AppData\Local\Temp\arduino_build_669581/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\nodemcu" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src" "C:\Users\Elliot\AppData\Local\Temp\arduino_build_669581\sketch\McLighting.ino.cpp" -o "nul"
"C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Elliot\AppData\Local\Temp\arduino_build_669581/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\nodemcu" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\DNSServer\src" "C:\Users\Elliot\AppData\Local\Temp\arduino_build_669581\sketch\McLighting.ino.cpp" -o "nul"
"C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/include" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/lwip2/include" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\Elliot\AppData\Local\Temp\arduino_build_669581/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536   -DARDUINO=10805 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU"  -DESP8266 "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\variants\nodemcu" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WiFi\src" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\DNSServer\src" "-IC:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266WebServer\src" "C:\Users\Elliot\AppData\Local\Temp\arduino_build_669581\sketch\McLighting.ino.cpp" -o [... cut ...]
Using library WS2812FX-master in folder: C:\Users\Elliot\Documents\Arduino\libraries\WS2812FX-master (legacy)
Using library Adafruit_NeoPixel at version 1.1.3 in folder: C:\Users\Elliot\Documents\Arduino\libraries\Adafruit_NeoPixel 
Using library Ticker at version 1.0 in folder: C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\Ticker 
Using library Hash at version 1.0 in folder: C:\Users\Elliot\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.0\libraries\Hash 
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

toblum, 2018-01-06:
Hi, @e11i0t23 

please have a look here: https://github.com/kitesurfer1404/WS2812FX/issues/58
It seems that the latest version of WS2812FX needs a newer ESP8266 core in some cases.

Regards
Tobias

e11i0t23, 2018-01-07:
re-installed esp8266 packages and all if fixed

closed 2018-01-07 by e11i0t23