toblum/McLighting #78: Problem z kompilacją

mlynekg, 2018-01-09:
Witam Serdecznie jestem tutaj nowy.
Mam taki problem z kompilacją:

Arduino:1.8.5 (Windows Store 1.8.10.0) (Windows 10), Płytka:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (1M SPIFFS)"

Opcje projektu zmienione, przeładuj całość
C:\Users\Grzegorz\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar: unable to rename 'C:\Users\Grzegorz\AppData\Local\Temp\arduino_build_808509/arduino.ar'; reason: File exists

exit status 1
Błąd kompilacji dla płytki NodeMCU 1.0 (ESP-12E Module).

Ten raport powinien zawierać więcej informacji jeśli w 
File -> Preferencje zostanie włączona opcja "Pokaż
szczegółowe informacje podczas kompilacji"

Jak to rozwiązać?

toblum, 2018-01-09:
Hi @mlynekg,

this may be due to problems with an antivirus software: https://forum.arduino.cc/index.php?topic=86547.0

Regards.
Tobias

mlynekg, 2018-01-09:
Dziękuję bardzo pomogło wyłączenie programu antywirusowego

Pozdrawiam

Grzegorz

closed 2018-01-09 by mlynekg