toblum/McLighting #79: Set hostname for DHCP

elmicha, 2018-01-11:
Hi,

thank you for the great firmware!

I found that the supplied DHCP hostname was ESP_XXXX (the mDNS hostname was ok). The browsers on my Android phone don't seem to know about mDNS, so http://mc_lighting.local didn't work.
After seeing https://github.com/tzapu/WiFiManager/issues/208 I tried 

    wifi_station_set_hostname(const_cast<char*>(HOSTNAME));

before the // Setup: WiFiManager comment, and now it request the correct hostname.

toblum, 2018-01-13:
Hi @elmicha,

thank you for pointing me to this. I added the line to the code.

Regards
Tobias

closed 2018-01-13 by toblum

brunoricardojava, 2018-01-23:
How do I access the browser? Can be hostname.local?

Lehmancreations, 2018-01-23:
If you have bonjour installed on your machine and your browser supports it.   I can get to it by hostname.local on my iPhone but not in Firefox.  I’m Firefox I have to find the ip and use that.

toblum, 2018-01-25:
@brunoricardojava There was a recent fix in McLIghting regarinf MDNS. Make sure to use the latest version.
But there are still some cases where you have to use the IP.