toblum/McLighting #80: MQTT does not work / resets device

T94T, 2018-01-12:
Hi,

First of all: Great project!

I have this setup running:
- ESP12F
- Latest McLighting repo
- Latest PubSubClient library
- Mosquitto Broker with authentication
- McLighting Hostname: McLighting

Everything works, but MQTT not. The console shows me:
> MQTT connected!
> MQTT topic in: McLighting/in
> MQTT topic out: ⸮?[ghting
Topic out seems really wrong to me.

Sending _=rainbow_ to topic _McLighting/in_ results in a wdt reset. The _mqtt_callback_ method does not even get called.

toblum, 2018-01-13:
Hi @T94T,

this is a bit strange as the in-topic looks OK. Does it work when you set the out-topic to a static value in McLighting.ino?

https://github.com/toblum/McLighting/blob/2956fc348a9fd7e58495b4cca9701d161cf0673a/Arduino/McLighting/McLighting.ino#L321
e.g. `snprintf(mqtt_outtopic, sizeof mqtt_outtopic, "mclighting/out", HOSTNAME);`
or: `mqtt_outtopic = "mclighting/out";`

Regards
Tobias

T94T, 2018-01-14:
Very strange...
I edited the definitions.h file directly and commented the _snprintf_ line for the _mqtt_outtopic_.
```
char mqtt_intopic[strlen(HOSTNAME) + 4];
char mqtt_outtopic[15] = "mclighting/out";

```
Now the console output looks like this. This time the input topic looks strange.

> MQTT topic in: ⸮?[ghting
> MQTT topic out: mclighting/out

If I hardcode the definitions.h like this

```
char mqtt_intopic[] = "mclighting/in";
char mqtt_outtopic[] = "mclighting/out";
```

the variable _mqtt_reconnect_retries_ is set to 1883 from the beginning on. This seems to be my MQTT port.

When setting _mqtt_reconnect_retries_ back to 0, I get stuck in a wdt reset loop. 

I'm working with Arduino IDE 1.9.0 Beta. After hours of fiddling around I tried to compile with 1.8.3. And everything works!

toblum, 2018-01-14:
Hi @T94T,

thank you for reporting. I haven't tried out 1.9.0 yet, I'm currently using 1.8.5 with ESP core 2.4.0 now. But I will check 1.9.0 soon to prevent future problems.

Regards
Tobias

szepnorbee, 2018-01-14:
Hi!
I wanted to use this awesome firmware with MQTT, but is not working for me. I use this with nodeRED. Compiled with Arduino 1.8.5. (esp8266 core2.4.0)

[image]

[image]

toblum, 2018-01-21:
What was the problem with MQTT? The configuration looks OK.

BTW: Thank you for your contribution.

szepnorbee, 2018-01-22:
If i publish my message (payload "=all" or "*FF0000") in kitchenLED/in topic, then nothing happening. MQTT broker is mosquitto on raspberryPI. Broker is working, tested with APP.

You are welcome! Not at all.

toblum, 2018-01-22:
So it connects correctly to MQTT and the Web Client Works? What does serial log say? 
When  MQTT connects, you should also See that there is a „McLighting ready“ Message in the out topic. 

Please also Check of there is maybe another client on your broker that Users the same Clientid. 

Mosquitto is ok. I also use it. 

Regards 
Tobias

szepnorbee, 2018-01-23:
It's working now. I have uncomented this: wifiManager.resetSettings();  and 
Previously i have defined all the settings (mqtt host,port) under the definitions.h because the wifiManager did not show me for the first time. I do not know why but it was connected automatically to my home network. (Maybe there was something in the eeprom)

toblum, 2018-01-25:
Hi @szepnorbee,

great that it works now.

Regards
Tobias

closed 2018-01-25 by toblum