toblum/McLighting #82: unabe to upload index.htm

sander816, 2018-01-13:
i have a esp01, flashing mcligting was pretty easy. I can rech the esp through its ipadress
but when i go to http://ip/upload and upload the index.htm i get a blank page.
When i go back to http://ip  i get the File Not Found again

what am i doing wrong?

toblum, 2018-01-13:
Hi @sander816,

it looks that the SPIFFS filesystem is not working correctly. That happens sometimes on the ESP01 as the memory is very limited. How did you set the "Flash size" option in Arduino IDE?

Regards
Tobias

sander816, 2018-01-14:
512k
I selected generic esp8266 module

toblum, 2018-01-14:
You need to choose a setting that reserves some space for SPIFFS, e.g. 512K (128K SPIFFS).

insanity67, 2018-01-16:
Same issue here, ESP8266-01 with 1MB Flash. Does not matter what size of SPIFFS I choose, upload is not working. Even reinstalled Arduino IDE (portable), redownloaded all libs and so on. 
Every time File Not Found on / page.

insanity67, 2018-01-16:
somehow upload "did work", but now I just get random garbage in browser:

`Content-Length: 2048 Connection: close ¥¥¥¥¢MD=¥¥¥¥Content-Type: text/html ¥¥¥¥¢[C0 User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/63.0.3239.132 Safari/537.36 Upgrade- Content-Length: 2048 Connection: close ¥¥¥¥¢MD=¥¥¥¥Content-Type: text/html ¥¥¥¥¢[C0 User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/63.0.3239.132 Safari/537.36 Upgrade- Content-Length: 2048 Connection: close ¥¥¥¥¢MD=¥¥¥¥Content-Type: text/html ¥¥¥¥¢[C0 User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/63.0.3239.132 Safari/537.36 Upgrade- Content-Length: 2048 Connection: close ¥¥¥¥¢MD=¥¥¥¥Content-Type: text/html ¥¥¥¥¢[C0 User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/ Content-Length: 2048 Connection: close ¥¥¥¥¢MD=¥¥¥¥Content-Type: text/html ¥¥¥¥¢[C0 User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/63.0.3239.132 Safari/537.36 Upgrade- Content-Length: 2048 Connection: close ¥¥¥¥¢MD=¥¥¥¥Content-Type: text/html ¥¥¥¥¢[C0 User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/63.0.3239.132 Safari/537.36 Upgrade- Content-Length: 2048 Connection: close ¥¥¥¥¢MD=¥¥¥¥Content-Type: text/html ¥¥¥¥¢ Content-Length: 2048 Connection: close ¥¥¥¥¢MD=¥¥¥¥Content-Type: text/html ¥¥¥¥¢[C0 User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/63.0.3239.132 Safari/537.36 Upgrade- Content-Length: 2048 Connection: close ¥¥¥¥¢MD=¥¥¥¥Content-Type: text/html ¥¥¥¥¢[C0 User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit`

insanity67, 2018-01-16:
With this tool I was able to upload the index.htm to the SPIFFS filesystem, and its working like a charm:
https://github.com/esp8266/arduino-esp8266fs-plugin

toblum, 2018-01-20:
Hi @insanity67,

it's unclear why it didn't work the standard way, good to know, that you make it work using the uploader tool!

Regards
Tobias

toblum, 2018-01-20:
@sander816 Does this work for you also?

closed 2018-01-20 by toblum

insanity67, 2018-01-20:
Maybe this is an ESP8266-01 problem only.

toblum, 2018-01-20:
Yes, it looks like it's related to the limited memory on the ESP-01.

sander816, 2018-01-21:
Yes, but with selecting 64k in arduino
No extra tool needed

toblum, 2018-01-22:
Thank you for the feedback,. I added that information to the wiki.