toblum/McLighting #84: MQTT Adafruit.io

Laggylarry, 2018-01-15:
Looking for some help getting the mqtt to connect to an adafruit.io feed. 
Anyone got this working??

toblum, 2018-01-20:
Hi @Laggylarry,

what exactly does not work? Problems with the MQTT connection itself? What does the serial log say?

There seem to be known problems using Adafruit.io with the pubsubclient library. Maybe you want to have a looke here:
https://github.com/knolleary/pubsubclient/issues/291
or:
https://github.com/knolleary/pubsubclient/issues/256

Regards
Tobias

The-Holgi, 2018-03-16:
Hallo,
thanks for McLighting.
It works great over the webinterface.
Now i want to controll it over my fhem server with mqtt.
MQTT (mosquito) run‘s on a raspberry pi, how must the topic shown to set the led stripe ?
An example where great /espname/in/....
Regards Holger

toblum, 2018-03-17:
@The-Holgi Have a look here: https://github.com/toblum/McLighting/wiki/MQTT-API

closed 2018-03-17 by toblum